zephyrproject-rtos/zephyr #27511: coverage: qemu platforms: sanitycheck generates many `unexpected eof` failures when enable coverage

wangnuannuan, 2020-08-12:
**Describe the bug**
When I run coverage test on qemu platforms(qemu_x86, arc_qemu_em) with sanitycheck, there are many `unexpected eof` failures. But the `handler.log`  shows that the testcases passed, and they also contains complete coverage data. 

Following are the failed testcases on `qemu_x86`:

```
ERROR   - qemu_x86                  samples/posix/eventfd/sample.posix.eventfd         FAILED: unexpected eof
ERROR   - qemu_x86                  tests/subsys/logging/log_immediate/logging.log_immediate FAILED: unexpected eof
ERROR   - qemu_x86                  tests/subsys/fs/fat_fs_dual_drive/filesystem.fat.dual_drive FAILED: unexpected eof
ERROR   - qemu_x86                  tests/net/udp/net.udp                              FAILED: unexpected eof
ERROR   - qemu_x86                  tests/net/trickle/net.trickle                      FAILED: unexpected eof
ERROR   - qemu_x86                  tests/net/tcp2/net.tcp2.simple                     FAILED: unexpected eof
ERROR   - qemu_x86                  tests/net/tcp/net.tcp.simple                       FAILED: unexpected eof
ERROR   - qemu_x86                  tests/lib/heap_align/lib.heap_align                FAILED: unexpected eof
ERROR   - qemu_x86                  tests/kernel/threads/no-multithreading/kernel.threads.no-multithreading FAILED: unexpected eof
ERROR   - qemu_x86                  tests/kernel/early_sleep/kernel.common.sleep       FAILED: unexpected eof
ERROR   - qemu_x86                  tests/bluetooth/uuid/bluetooth.gatt                FAILED: unexpected eof
ERROR   - qemu_x86                  tests/crypto/rand32/crypto.rand32                  FAILED: unexpected eof
ERROR   - qemu_x86                  tests/crypto/rand32/crypto.rand32.random_sw_systimer FAILED: unexpected eof
ERROR   - qemu_x86                  tests/drivers/console/drivers.console.uart         FAILED: unexpected eof
ERROR   - qemu_x86                  tests/drivers/eeprom/peripheral.eeprom             FAILED: unexpected eof
ERROR   - qemu_x86                  tests/drivers/counter/counter_cmos/drivers.counter.cmos FAILED: unexpected eof
ERROR   - qemu_x86                  tests/drivers/flash_simulator/drivers.flash.flash_simulator FAILED: unexpected eof
ERROR   - qemu_x86                  tests/drivers/ipm/drivers.ipc.mailbox              FAILED: unexpected eof
ERROR   - qemu_x86                  tests/drivers/sensor/driver.sensor                 FAILED: unexpected eof
ERROR   - qemu_x86                  tests/drivers/hwinfo/api/drivers.device_id         FAILED: unexpected eof
ERROR   - qemu_x86                  tests/benchmarks/timing_info/benchmark.kernel.timing.default FAILED: unexpected eof
ERROR   - qemu_x86                  tests/benchmarks/sys_kernel/benchmark.kernel.core  FAILED: unexpected eof
ERROR   - qemu_x86                  tests/application_development/gen_inc_file/buildsystem.include_file FAILED: unexpected eof
ERROR   - qemu_x86                  tests/arch/x86/boot_page_table/kernel.memory_protection.x86.bootpage.validation FAILED: unexpected eof
ERROR   - qemu_x86                  tests/arch/x86/static_idt/arch.interrupt.x86       FAILED: unexpected eof
ERROR   - qemu_x86                  tests/arch/x86/x86_mmu_api/kernel.memory_protection.mmu FAILED: unexpected eof
```

andrewboie, 2020-08-12:
Wondering if https://github.com/zephyrproject-rtos/zephyr/issues/26949 has something to do with this, since coverage data spams the log?

nashif, 2020-09-10:
duplicate of  #26949

closed 2020-09-10 by nashif

smrtos, 2022-07-30:
[image]

Just saw some similar `EOF` error with `qemu_riscv32_smp`. It also involves logging and printk.