zephyrproject-rtos/zephyr #27517: FCB limited to flash areas of 255 sectors

chris-krogsystems, 2020-08-12:
The Flash Circular Buffer is currently limited to flash areas of 255 sectors or less. For flash with a standard 4K sector, this equates to slightly less than 1MB. With external flash and some long-term store-and-forward applications, a larger FCB is required.

The limit, from what I can see, is caused by the sector_cnt field in struct fcb being of type u8_t. Changing this to a u32_t fixes the issue, and does not cause any side-effects that I can see so far in my limited testing.

A u16_t would allow for almost 256MB storage, but I would recommend a u32_t, as 1GB of measurement storage does not seem to be outside the realm of possibility in the next few years.

p-sousa, 2025-02-04:
@chris-krogsystems did you ended up updating the `sector_cnt` field in your project?

I am considering using a 16MB NOR Flash with 4K sectors (iamge at the end of mem. organization), and I wonder if changing `sector_cnt` from `u8_t` to `u32_t`, as suggested by @chris-krogsystems, makes sense in this context. Are there any potential limitations or breaking changes that should be considered for existing Zephyr users?

In https://github.com/zephyrproject-rtos/zephyr/pull/34013 @Laczen suggests increasing the sector size to a multiple of 4K instead, but in my project, this creates two challenges:
* To reduce wear on the flash, I buffer multiple 16/32-byte entries in RAM before committing them to flash. With 4K sectors, this already consumes a significant amount of RAM. Increasing the sector size further (e.g., to 64K) would significantly increase RAM usage. This makes 4K sectors a lot more desirable.
*  In my application, retaining as much data as possible after the FCB fills up is ideal. Since FCB erases an entire sector when full, a 4K sector aligns perfectly with my flash hardware and minimizes data loss per erase cycle.

@de-nordic @carlescufi @nvlsianpu @Laczen @avkanez, your feedback would be greatly appreciated (mentioning you since you're involved in the issue). Please feel free to correct any of my assumptions.

I’d be happy to work on a PR for this if the change makes sense. Thanks in advance!

[image]

avkanez, 2025-02-05:
Hi @p-sousa,
I'm still using `u32_t sector_cnt` in my project (applying a patch to Zephyr whenever I migrate to a newer SDK). Everything works like a charm.

p-sousa, 2025-02-10:
Thank you for the input @avkanez !

Is there interest in having this in upstream? Even if it is through a Kconfig option?

Ty all.

AdrianTayag, 2025-09-25:
Just chiming in since I am also applying the patch to increase sector counts for my use case.

This change only requires a change of sector_cnt.  Not sure what holds this enhancement request to be not be implemented?