zephyrproject-rtos/zephyr #27522: shell: Output can get corrupted when printing from thread before command completes

nordic-krch, 2020-08-12:
**Describe the bug**
It has been seen that if command is executed which triggers another thread and that thread writes to shell then output may get corrupted. It may happen if thread prints before shell cleans up after the command (print another prompt). As the outcome interleaved output can be seen as below where `Done` message comes from external thread.

```
uart:~$ zdo active_ep 0x0000
src_addr=0 ep=242 
uart:~$ zdoDone
uart:~$
```

**To Reproduce**
Issue can be reproduced on out of tree example in NCS.

**Expected behavior**

expected output:
```
uart:~$ zdo active_ep 0x0000
src_addr=0 ep=242 
Done
uart:~$
```

**Impact**
User experience.

de-nordic, 2020-08-12:
This is common problem for multithreading systems when same tty is used as stdout for different threads, e.g. executing in bash shell on Linux/Freebsd:
```
while true; do sleep 1; echo -n hello; done &
```

will interrupt prompt by printing `hello` every second.
```
prompt:
prompt: while true; do sleep 1; echo -n hello; done &
[1] 142
prompt: hellohellohello
prompt:
prompt: hellokilhellol hello%1
prompt:
[1]+  Terminated              while true; do
    sleep 1; echo -n hello;
done
prompt:
```

This can not be easily fixed as it would require detaching console from other threads when shell is active.
Some solution might be creation of virtual console output buffer, where an output from background threads would be redirected,  that would be accessed via virtual file or special command.

jakub-uC, 2020-08-12:
This is expected. Shell supports printing from command context and logger context. Any other usage of shell print commands is not synchronized. If other thread want to print something I recommend to use log module. Otherwise, we would need to create another log like object.

nordic-krch, 2020-08-13:
> This is common problem for multithreading systems

> This is expected.

It is expected that one thread may preempt another and interleave data but here the problem is a bit different. Command is executed here:
```
k_mutex_unlock(&shell->ctx->wr_mtx);
flag_cmd_ctx_set(shell, 1);
ret_val = shell->ctx->active_cmd.handler(shell, argc,
							 (char **)argv);
flag_cmd_ctx_set(shell, 0);
/* Bring back mutex to shell thread. */
k_mutex_lock(&shell->ctx->wr_mtx, K_FOREVER);
```
Mutex is temporary unlocked to allow `shell_print` from the handler. Also context is set and then unset. Then after command execution `state_set(shell, SHELL_STATE_ACTIVE);` which clear the buffer and prints the prompt. It seems to me that if external thread preempts and prints somewhere between that process then this corruption may happen and maybe it can be prevented contrary to the case when one context preempts another to print to shell.

jakub-uC, 2020-08-17:
@nordic-krch Reason for such implementation was a discussion about deadlock: #13260.

Regarding your assumption I am not sure if they are correct. When we are in this area of code:
```
k_mutex_unlock(&shell->ctx->wr_mtx);
flag_cmd_ctx_set(shell, 1);
ret_val = shell->ctx->active_cmd.handler(shell, argc,
							 (char **)argv);
flag_cmd_ctx_set(shell, 0);
/* Bring back mutex to shell thread. */
k_mutex_lock(&shell->ctx->wr_mtx, K_FOREVER);
```

from the display perspective, the cursor is at the beginning of a new and empty line and there is no prompt.
The only thing that `state_set(shell, SHELL_STATE_ACTIVE);` will do in regards to the terminal is printing the prompt. 
So after calling `state_set` I would expect error like: `Doneuart:~$ `

At this point (mutex is unlocked), prints form the shell command and other threads will interleave. 

I would need to reproduce the problem to move forward with the solution.
I have one additional question about the line: `uart:~$ zdoDone`
Did you start typing `zdo` and `Done` appeared or?

PR #27926 (2020-09-03): shell: fix printing from other threads
This fix prevents a scenario when command callback calls shell_fprintf
after the shell cleared the command context flag and before setting
the prompt correctly.

Fixes #27522
commit 5554bdca (2020-09-03): shell: fix printing from other threads

This fix prevents a scenario when command callback calls shell_fprintf
after the shell cleared the command context flag and before setting
the prompt correctly.

Fixes #27522
closed 2020-09-03 by carlescufi