zephyrproject-rtos/zephyr #27526: poll(2) returning -1 errno ENOMEM

cfriedt, 2020-08-12:
**Describe the bug**
I think I may have found a regression in poll(2).

When z_fdtable_call_ioctl is called, and it returns -ENOMEM, I believe that is simply used to indicate that all of the poll fds have been prepared, which is technically not an error.

The result is that poll returns -1 with errno set to ENOMEM

CC: @jukkar 
**To Reproduce**

**Expected behavior**

Poll should block, not return -1 and not set errno to ENOMEM

**Impact**

Could have significant impact to a number of users.

**Logs and console output**

**Additional context**

 diff --git a/subsys/net/lib/sockets/sockets.c b/subsys/net/lib/sockets/sockets.c
 index e39545ca8f..27731de565 100644
 --- a/subsys/net/lib/sockets/sockets.c
 +++ b/subsys/net/lib/sockets/sockets.c
 @@ -1239,6 +1239,11 @@ int z_impl_zsock_poll(struct zsock_pollfd *fds, int nfds, int poll_timeout)
                         return z_fdtable_call_ioctl(vtable, ctx,
                                                     ZFD_IOCTL_POLL_OFFLOAD,
                                                     fds, nfds, poll_timeout);
 +               } else if (result == -ENOMEM) {
 +                       /* If POLL_PREPARE returned ENOMEM, it means
 +                        * all of the poll fds have been prepared.
 +                        */
 +                       break;
                 } else if (result != 0) {
                         errno = -result;
                         return -1;

jukkar, 2020-08-12:
@cfriedt could you send a PR for your proposal fix

cfriedt, 2020-08-13:
@jukkar I likely will not have any time in the next couple of weeks.

andrewboie, 2020-08-13:
@jukkar I asked @cfriedt to file a bug on this in slack as he didn't have time to prepare a fix, just noticed it when doing something else

jukkar, 2020-08-14:
No worries, I can prepare a PR for this.

jukkar, 2020-08-14:
I started to check this by first creating a unit test to the expected behavior. If I am reading the code right, the ENOMEM is only returned if CONFIG_NET_SOCKETS_POLL_MAX (default value 3) is smaller that what user tries to feed to it (nfds parameter). In practice this would mean that poll() would not be able to properly check the file descriptors as some of the user fed fds would never be checked. To me this looks like a valid restriction.

Could you elaborate how you tested this?

cfriedt, 2020-08-14:
I think you're right. I may have misinterpreted it. Will close

closed 2020-08-14 by cfriedt