zephyrproject-rtos/zephyr #27544: TrustZone: NSC_ALIGN gets redefined

INTENDRO, 2020-08-13:
**Describe the bug**
`NSC_ALIGN` gets redefined in the file `zephyr/arch/arm/core/aarch32/cortex_m/tz/secure_entry_functions.ld` when the config `CONFIG_CPU_HAS_NRF_IDAU` is set and the config `CONFIG_ARM_NSC_REGION_BASE_ADDRESS` is set to a value not equal to zero. Up until v2.2.1, the file was structured differently and the symbol did not get redefined. Zephyr v2.3.0 added changes in order to support the nRF53 chip which introduced the possibility for this redefinition.

**To Reproduce**
Steps to reproduce the behavior:
1. Setup Nordic Connect SDK
2. Open a sample using TrustZone
3. Use a cpu with an nRF IDAU. E.g. nRF9160 DK
4. Set `CONFIG_ARM_NSC_REGION_BASE_ADDRESS` to a value not equal to zero. E.g. 0x0001ffc0
5. west build -b nrf9160dk_nrf9160ns
6. See warning

**Expected behavior**
NSC_ALIGN should only get defined once.

**Impact**
Produces unwanted warning output during compilation (annoyance).

**Logs and console output**
```
[48/195] Generating linker.cmd
In file included from /home/nosd/github/trustzone_test/04_ns_function_call/non_secure_app/build/spm/zephyr/include/generated/snippets-sections.ld:3,
                 from /home/nosd/github/ncs/zephyr/include/arch/arm/aarch32/cortex_m/scripts/linker.ld:509,
                 from /home/nosd/github/ncs/zephyr/soc/arm/nordic_nrf/nrf91/linker.ld:9:
/home/nosd/github/ncs/zephyr/include/../arch/arm/core/aarch32/cortex_m/tz/secure_entry_functions.ld:18: warning: "NSC_ALIGN" redefined
   18 |  #define NSC_ALIGN . = ABSOLUTE(CONFIG_ARM_NSC_REGION_BASE_ADDRESS)
      | 
/home/nosd/github/ncs/zephyr/include/../arch/arm/core/aarch32/cortex_m/tz/secure_entry_functions.ld:10: note: this is the location of the previous definition
   10 |  #define NSC_ALIGN \
      | 
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: gnuarmemb-9-2019-q4-major
 - Tag: Nordic Connect SDK Version 1.3.0

**Additional context:**
I assume that the support for nRF53 can be guaranteed and still have no redenition of the symbol whan using the nRF91. However, I just started using TrustZone features and I am unsure if my assumption is correct.

ioannisg, 2020-09-14:
@oyvindronningstad  will you be able to provide the fix for this issue (probably present in #27872, which is still a draft) :)

oyvindronningstad, 2020-09-16:
> 
> 
> @oyvindronningstad will you be able to provide the fix for this issue (probably present in #27872, which is still a draft) :)

Working on it now

PR #27872 (2020-09-23): arm: tz: secure_entry_functions.ld: Fix NSC_ALIGN redefinition
Allow CONFIG_ARM_NSC_REGION_BASE_ADDRESS to override the nRF-specific
logic for alignment.

Fixes #27544
closed 2020-09-23 by ioannisg