zephyrproject-rtos/zephyr #27548: CMake and west doesn't accept multiple overlay files during build

maksimmasalski, 2020-08-13:
**Describe the bug**
According to that documentation https://docs.zephyrproject.org/latest/guides/dts/howtos.html#set-devicetree-overlays I can submit multiple files. When I tried to do that for the mec15xxevb_assy6853 board for the test `/tests/drivers/gpio/gpio_basic_api` I faced with the fact, that command detects the first file, second not detected.

**To Reproduce**
1. Go to `/tests/drivers/gpio/gpio_basic_api/boards`
2. Create one more copy of the `mec15xxevb_assy6853.overlay` file and rename it like `2mec15xxevb_assy6853.overlay`
3. Go back to `/tests/drivers/gpio/gpio_basic_api/`
4. Run west using semicolon to separate files:
```
west build -p -b mec15xxevb_assy6853 . -DDTC_OVERLAY_FILE=./boards/mec15xxevb_assy6853.overlay;./boards/2mec15xxevb_assy6853.overlay 
```
In output you will see that build was successful, but only 1 overlay file was detected.

5. Try west using space to separate files:
```
west build -p -b mec15xxevb_assy6853 . -DDTC_OVERLAY_FILE=./boards/mec15xxevb_assy6853.overlay ./boards/2mec15xxevb_assy6853.overlay
```
In output you will see that build was successful, but only 1 overlay file was detected.

6. Try CMake. In documentation is written that I should use semicolon for cmake using command:
```
cmake -GNinja -DBOARD=mec15xxevb_assy6853 -DDTC_OVERLAY_FILE=/home/maksim/zephyrproject/zephyr/tests/drivers/gpio/gpio_basic_api/boards/mec15xxevb_assy6853.overlay;/home/maksim/zephyrproject/zephyr/tests/drivers/gpio/gpio_basic_api/boards/2mec15xxevb_assy6853.overlay ..
```

<b>Error output (means documentation is wrong and semicolon is not working)</b>
```
[maksim@maksim-nuc gpio_basic_api]$ cmake -GNinja -DBOARD=mec15xxevb_assy6853 -DDTC_OVERLAY_FILE=/home/maksim/zephyrproject/zephyr/tests/drivers/gpio/gpio_basic_api/boards/mec15xxevb_assy6853.overlay;/home/maksim/zephyrproject/zephyr/tests/drivers/gpio/gpio_basic_api/boards/2mec15xxevb_assy6853.overlay ..
CMake Warning:
  No source or binary directory provided.  Both will be assumed to be the
  same as the current working directory, but note that this warning will
  become a fatal error in future CMake releases.

Including boilerplate (Zephyr base): /home/maksim/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/maksim/zephyrproject/zephyr/tests/drivers/gpio/gpio_basic_api
-- Zephyr version: 2.3.99 (/home/maksim/zephyrproject/zephyr)
-- Found Python3: /usr/bin/python3.6 (found suitable exact version "3.6.8") found components:  Interpreter
-- Found west: /home/maksim/.local/bin/west (found suitable version "0.7.2", minimum required is "0.7.1")
CMake Error at /home/maksim/zephyrproject/zephyr/cmake/app/boilerplate.cmake:132 (message):
  Source directory equals build directory.  In-source builds are not
  supported.  Please specify a build directory, e.g.  cmake -Bbuild -H.
Call Stack (most recent call first):
  /home/maksim/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /home/maksim/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:35 (include_boilerplate)
  CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
-bash: /home/maksim/zephyrproject/zephyr/tests/drivers/gpio/gpio_basic_api/boards/2mec15xxevb_assy6853.overlay: Permission denied
```
7. Try CMake with space to separate 2 overlay files. 
```
cmake -GNinja -DBOARD=mec15xxevb_assy6853 -DDTC_OVERLAY_FILE=/home/maksim/zephyrproject/zephyr/tests/drivers/gpio/gpio_basic_api/boards/mec15xxevb_assy6853.overlay /home/maksim/zephyrproject/zephyr/tests/drivers/gpio/gpio_basic_api/boards/2mec15xxevb_assy6853.overlay ..
```
<b>Result</b>
Detected only 1 file.

**Expected behavior**
I expect to give to west or cmake my overlay files and they all must be detected and Info messages must be visible in the terminal output for the each file.

**Impact**
I can't test many gpios on the Microchip if I have only 1 overlay file, possible solution to create 1 overlay file for all gpios, but in documentation is stated that I can add multiple files.

**Environment (please complete the following information):**
 - OS: Linux Fedora 28
 - Toolchain 0.11.4
 - Commit SHA ca99404832f9053d2187f0994509e24a17199956

nashif, 2020-08-13:
` west build -p -b mec15xxevb_assy6853 -- -DDTC_OVERLAY_FILE="boards/mec15xxevb_assy6853.overlay boards/2mec15xxevb_assy6853.overlay"`

```
-- Found dtc: /home/nashif/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/nashif/Work/zephyrproject/zephyr/boards/arm/mec15xxevb_assy6853/mec15xxevb_assy6853.dts
-- Found devicetree overlay: boards/mec15xxevb_assy6853.overlay
-- Found devicetree overlay: boards/2mec15xxevb_assy6853.overlay
```

it works as expected...

maksimmasalski, 2020-08-13:
@nashif Thanks, I will modify docs file, because in docs no quotations " " in that command

`on the cmake build command line (-DDTC_OVERLAY_FILE=file1.overlay;file2.overlay)`

tejlmand, 2020-08-13:
Please keep the `;`, as that is the correct CMake way of specifying a list.
That is:
> on the cmake build command line (-DDTC_OVERLAY_FILE="file1.overlay;file2.overlay")

tejlmand, 2020-08-13:
The problem is actually the shell, cause most shells interprets the `;` as a command separator.

So escaping the `;` is also an option (at least in `bash` and `fish`): 
```
-DDTC_OVERLAY_FILE="file1.overlay\;file2.overlay"
```

maksimmasalski, 2020-08-13:
Thanks for the comments, will keep ; and will add semicolon in the docs examples.

closed 2020-08-13 by nashif

avigreen1978, 2020-09-23:
Can you please demonstrate how to do that in CMakeLists.txt `DTC_OVERLAY_FILE` variable?
I have tried those combinations:
`set(DTC_OVERLAY_FILE "file1.overlay;file2.overlay")`
`set(DTC_OVERLAY_FILE "file1.overlay" "file2.overlay")`
`set(DTC_OVERLAY_FILE file1.overlay;file2.overlay)`
`set(DTC_OVERLAY_FILE file1.overlay file2.overlay)`
And none of them work.
Thanks

tejlmand, 2020-09-23:
> with this line:
> `string(REPLACE " " ";" DTC_OVERLAY_FILE_AS_LIST "${DTC_OVERLAY_FILE}")`
> (Adding quotation marks around `${DTC_OVERLAY_FILE}`).
> Tell me if you find any better way, and if not, I'll open a PR

That is the correct way, and also the way used for `CONF_FILE` and `SHIELD`.
Not sure why that wasn't been used in this case.

So feel free to open a PR.

A couple of additional notes:
> I have tried those combinations:

This should have worked:
```
 set(DTC_OVERLAY_FILE "file1.overlay file2.overlay")
```

when filing the PR, then I would suggest you  use:
```
list(APPEND DTC_OVERLAY_FILE <file>)
```
in your own sample, to still allow users to also specify overlay files on command line.