zephyrproject-rtos/zephyr #27558: "west update" only certain vendor

ahmedwahdan, 2020-08-13:
How to use "west update" to update only certain vendor related files, i.e stm32

galak, 2020-08-13:
You need to modify the manifest (west.yml) to slim it down to just what you need.

galak, 2020-08-13:
@mbolivar-nordic (adding if you've got any other comments or doc pointers)

mbolivar-nordic, 2020-08-13:
I don't understand the question:

- If you're talking about limiting the set of repositories in the manifest, I would use manifest imports with a list that filters out just want you want. See https://docs.zephyrproject.org/latest/guides/west/manifest.html for information and many examples.
- If you're talking about just updating some project, then give that project name to `west update`, like `west update hal_stm32`. See `west help update` for help on the command.

closed 2020-08-13 by mbolivar-nordic