zephyrproject-rtos/zephyr #27559: How to use stm32cubeIDE to build and debug?

ahmedwahdan, 2020-08-13:
How to use stm32cubeIDE to build and debug?

erwango, 2020-08-24:
@ahmedwahdan stm32cubeIDE is not made to use with zephyr. If you need an IDE to use zephyr I suggest to use VSCode (cf https://github.com/zephyrproject-rtos/zephyr/issues/21119 for configuration) or Eclipse with Zephyr plugin (http://marketplace.eclipse.org/content/eclipse-plug-zephyr-project).

erwango, 2020-09-03:
@ahmedwahdan are you satisfied with the answer provided ? 
If yes, can we close the issue?

EminYagmahan, 2020-09-11:
Alternatively I can recommend CLion from JetBrains. With the Embedded Development Support Plugin, STM32CubeMX and OpenOCD can be integrated into the IDE quite well. This is how it can look like. Zephyr projects also work great. A guide is available here:
https://www.jetbrains.com/clion/features/embedded-dev.html
[image]

erwango, 2020-09-14:
Thanks @EminYagmahan.
I would just warn people that using Zephyr, you won't be able to use the configuration and code generated by CubeMx.

closed 2020-11-28 by github-actions[bot]