zephyrproject-rtos/zephyr #27566: nRF52832: MCUBoot cannot read signed SMP Server Sample binary

kyberdin, 2020-08-13:
**Description**
While following the [SMP Server Sample](https://docs.zephyrproject.org/latest/samples/subsys/mgmt/mcumgr/smp_svr/README.html#smp-svr-sample) for an `nRF52dk_nrf52832` board, flashing the signed binary results in output from MCUBoot that it can't read the image headers and that the image is invalid.

**To Reproduce:**
My commands, in order with the guide steps:
1. `west build -b nrf52dk_nrf52832 -d build_mcuboot bootloader/mcuboot/boot/zephyr`
2. `west flash -d build_mcuboot`
3. `west build -b nrf52dk_nrf52832 zephyr/samples/subsys/mgmt/mcumgr/smp_svr -- -DOVERLAY_CONFIG=overlay-bt.conf`
4. `west sign -t imgtool -- --key bootloader/mcuboot/root-rsa-2048.pem`
5. `west flash --bin-file build/zephyr/zephyr.signed.bin `

All of the above compile and flash successfully.

**Expected behavior**
Image would be validated and ready for communicating with `mcumgr`.

**Impact**
Showstopper. Cannot proceed with DFU process over BLE.

**Logs and console output**
Bootloader output:
```
*** Booting Zephyr OS build zephyr-v2.3.0-1818-g07b4df1945b5  ***
[00:00:00.004,913] ␛[0m<inf> mcuboot: Starting bootloader␛[0m
[00:00:00.011,108] ␛[0m<inf> mcuboot: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3␛[0m
[00:00:00.021,514] ␛[0m<inf> mcuboot: Boot source: none␛[0m
[00:00:00.027,130] ␛[1;33m<wrn> mcuboot: Failed reading image headers; Image=0␛[0m
[00:00:00.034,820] ␛[1;31m<err> mcuboot: Image in the primary slot is not valid!␛[0m
[00:00:00.042,663] ␛[1;31m<err> mcuboot: Unable to find bootable image␛[0m
```

**Environment:**
 - OS: Ubuntu 18.04.5
 - Zephyr 2.3.0, Zephyr SDK 11.3
 - SHAs:
    - `mcuboot`: `813d29c`
    - `zephyr`: `07b4df1`

**Additional context**
* `imgtool` (**1.6.0**) installed through `pip install imgtool` while in a Python3.6 virtualenv consisting of Zephyr + MCUBoot requirement packages.
* Signed binary flashed by `nrfjprog`, automatically pin resetting the device after flash.
* Building from the general `zephyrproject` setup resulting from the Getting Started guide; not a custom project.

de-nordic, 2020-08-14:
Can you post here shell output of the
```
west flash --bin-file build/zephyr/zephyr.signed.bin 
```
command?
Try using:
```
west flash --hex-file build/zephyr/zephyr.signed.hex
instead, is it OK?

kyberdin, 2020-08-14:
```
$ west flash --bin-file build/zephyr/zephyr.signed.bin
-- west flash: rebuilding
[0/1] cd /home/saruman/zephyrproject/build/zephyr/cmake/flash && /usr/bin/cmake -E echo

-- west flash: using runner nrfjprog
-- runners.nrfjprog: Flashing file: /home/saruman/zephyrproject/build/zephyr/zephyr.hex
Parsing hex file.
Erasing page at address 0xC000.
Erasing page at address 0xD000.
Erasing page at address 0xE000.
Erasing page at address 0xF000.
Erasing page at address 0x10000.
Erasing page at address 0x11000.
Erasing page at address 0x12000.
Erasing page at address 0x13000.
Erasing page at address 0x14000.
Erasing page at address 0x15000.
Erasing page at address 0x16000.
Erasing page at address 0x17000.
Erasing page at address 0x18000.
Erasing page at address 0x19000.
Erasing page at address 0x1A000.
Erasing page at address 0x1B000.
Erasing page at address 0x1C000.
Erasing page at address 0x1D000.
Erasing page at address 0x1E000.
Erasing page at address 0x1F000.
Erasing page at address 0x20000.
Erasing page at address 0x21000.
Erasing page at address 0x22000.
Erasing page at address 0x23000.
Erasing page at address 0x24000.
Erasing page at address 0x25000.
Erasing page at address 0x26000.
Erasing page at address 0x27000.
Erasing page at address 0x28000.
Erasing page at address 0x29000.
Erasing page at address 0x2A000.
Erasing page at address 0x2B000.
Erasing page at address 0x2C000.
Erasing page at address 0x2D000.
Erasing page at address 0x2E000.
Erasing page at address 0x2F000.
Erasing page at address 0x30000.
Erasing page at address 0x31000.
Erasing page at address 0x32000.
Erasing page at address 0x33000.
Erasing page at address 0x34000.
Erasing page at address 0x35000.
Erasing page at address 0x36000.
Erasing page at address 0x37000.
Erasing page at address 0x38000.
Erasing page at address 0x39000.
Applying system reset.
Checking that the area to write is not protected.
Programming device.
Enabling pin reset.
Applying pin reset.
-- runners.nrfjprog: Board with serial number 682178881 flashed successfully.
```

Looks like flashing the hex file works just fine:

```
*** Booting Zephyr OS build zephyr-v2.3.0-1818-g07b4df1945b5  ***
[00:00:00.004,913] ␛[0m<inf> mcuboot: Starting bootloader␛[0m
[00:00:00.011,108] ␛[0m<inf> mcuboot: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3␛[0m
[00:00:00.021,514] ␛[0m<inf> mcuboot: Boot source: none␛[0m
[00:00:00.027,191] ␛[0m<inf> mcuboot: Swap type: none␛[0m
[00:00:00.298,950] ␛[0m<inf> mcuboot: Bootloader chainload address offset: 0xc000␛[0m
[00:00:00.306,945] ␛[0m<inf> mcuboot: Jumping to the first image slot␛[0m
*** Booting Zephyr OS build zephyr-v2.3.0-1818-g07b4df1945b5  ***

␛[1;32muart:~$ ␛[m␛[8D␛[J[00:00:00.152,313] ␛[0m<inf> smp_bt_sample: Advertising successfully started␛[0m
␛[1;32muart:~$ ␛[m␛[8D␛[J␛[1;32muart:~$ ␛[m
```

de-nordic, 2020-08-14:
Yes, seems that there is some problem with `--bin-file` switch.

kyberdin, 2020-08-14:
@de-nordic Noticed in the two west outputs of `--bin-file` and `--hex-file`:

Bin:
```
$ west flash --bin-file build/zephyr/zephyr.signed.bin
-- west flash: rebuilding
[0/1] cd /home/saruman/zephyrproject/build/zephyr/cmake/flash && /usr/bin/cmake -E echo

-- west flash: using runner nrfjprog
-- runners.nrfjprog: Flashing file: /home/saruman/zephyrproject/build/zephyr/zephyr.hex
```

Hex:
```
west flash --hex-file build/zephyr/zephyr.signed.hex
-- west flash: rebuilding
[0/1] cd /home/saruman/zephyrproject/build/zephyr/cmake/flash && /usr/bin/cmake -E echo

-- west flash: using runner nrfjprog
-- runners.nrfjprog: Flashing file: build/zephyr/zephyr.signed.hex
```

I would expect the `--bin-file` to flash the `zephyr.signed.bin` and not the general `zephyr.hex` output, like `--hex-file` does with `zephyr.signed.hex`

de-nordic, 2020-08-14:
@kyberdin the nrfjprog command, that is invoked by west flash to program nrf52840, does not support uploading bin files.
You will need to use `--hex-file` parameter with hex file as an argument.

kyberdin, 2020-08-14:
@de-nordic Will do! Thanks for the info and assist.

de-nordic, 2020-08-14:
@kyberdin One more info: when you will be using mcumgr to upload files via serial/bluetooth use bin files instead, not hex files.

kyberdin, 2020-08-14:
Thanks, I'll keep that in mind. Speaking of which, my problem has moved past `west` and is now with `mcumgr`:

```
sudo mcumgr --conntype ble --connstring ctlr_name=hci0,peer_name='Zephyr' image upload build/zephyr/zephyr.signed.bin
 0 B / 182.20 KiB [-----------------------------------------------------------------------------------------------------------------------------------------------------]   0.00%
panic: send on closed channel

goroutine 596 [running]:
mynewt.apache.org/newtmgr/nmxact/nmp.(*Listener).AfterTimeout.func1()
	/home/saruman/go/src/mynewt.apache.org/newtmgr/nmxact/nmp/dispatch.go:50 +0x65
created by time.goFunc
	/usr/local/go/src/time/sleep.go:167 +0x45
```

Application doesn't offer much help:
```
␛[1;32muart:~$ ␛[m␛[8D␛[J[00:12:35.377,197] ␛[0m<inf> smp_bt_sample: Advertising successfully started␛[0m
␛[1;32muart:~$ ␛[m␛[8D␛[J␛[1;32muart:~$ ␛[m␛[8D␛[J[00:15:59.451,721] ␛[0m<inf> smp_bt_sample: Connected␛[0m
␛[1;32muart:~$ ␛[m␛[8D␛[J␛[1;32muart:~$ ␛[m␛[8D␛[J[00:16:02.644,012] ␛[0m<inf> smp_bt_sample: Disconnected (reason 0x13)␛[0m
␛[1;32muart:~$ ␛[m␛[8D␛[J[00:16:02.644,958] ␛[0m<inf> smp_bt_sample: Advertising successfully started␛[0m
␛[1;32muart:~$ ␛[m␛[8D␛[J␛[1;32muart:~$ ␛[m
```

Though `btmon` is quite active during the attempt.

Other `mcumgr` commands still work though, such as `image list`:
```
sudo mcumgr --conntype ble --connstring ctlr_name=hci0,peer_name='Zephyr' image list                                 
Images:
 image=0 slot=0
    version: 0.0.0
    bootable: true
    flags: active confirmed
    hash: fea488b32bd50d104c79b805d52912452cfaf2aad46442afdd410deb181acb25
Split status: N/A (0)
```

Might be out of scope for this bug report though. I can open a new ticket if I'm unable to resolve it on my own.

de-nordic, 2020-08-14:
@kyberdin Please open other issue for that problem if it persist. I have not been able to reproduce the problem with `closed channel`

kyberdin, 2020-08-14:
Looks like the issue with `mcumgr` has [already been addressed](https://github.com/zephyrproject-rtos/zephyr/issues/27331). Closing the issue from @de-nordic 's feedback.

closed 2020-08-14 by kyberdin