zephyrproject-rtos/zephyr #27569: samples.usb.cdc-acm-composite failed on mimxrt1050_evk board

Zhaoningx, 2020-08-14:
Describe the bug
building samples/subsys/usb/cdc_acm_composite/ board, it would report a wrong.

To Reproduce
Steps to reproduce the behavior:

    west build -b mimxrt1050_evk samples/subsys/usb/cdc_acm_composite/
    west flash
    see error

Logs and console output
DEBUG   - DEVICE: *** Booting Zephyr OS build zephyr-v2.3.0-1774-g34859d0d3e40  ***
DEBUG   - DEVICE: [00:00:00.005,000] <err> usb_dc_mcux_ehci: Wrong endpoint index/address
DEBUG   - DEVICE: [00:00:00.005,000] <err> cdc_acm_composite: Failed to enable USB

Environment (please complete the following information):

    OS: Fedora28
    Toolchain: Zephyr-sdk-0.11.3
    Commit ID: 34859d0d3e

dleach02, 2020-08-19:
Where things stopped working is with e0b2227c, but looking at the PR it appears to add tests for (ep_abs_idx >= NUM_OF_EP_MAX) so I suspect the USB data structures always reported a value larger than NUM_OF_EP_MAX. So still digging.

dleach02, 2020-08-20:
@jfischer-phytec-iot, what I'm seeing is that the way the kinetis and mcux implementations of `usb_dc_ep_check_cap()` is doing a test on BIT(0) of the index value to determine if the endpoint can be an IN endpoint. The higher layer logic appears to be trying to put the IN endpoints at address 0x80 where the 0x80 bit mask indicates an IN endpoint. A macro extracts the true index value using:

n*2 + 1 (if the 0x80 bit is set)

This macro basically ensures that the IN endpoints will be on odd index boundaries while the OUT endpoints will be on even index boundaries. But this usb_dc_ep_check_cap() probably doesn't want to make this assumption and instead let the calling function, usb_validate_ep_cfg_data() sequentially set the index without concern about the even/odd.

But the problem is that the usb_dc_ep_check_cap() is checking a temporary address value and returning false because it knows it is an IN endpoint but the address is not odd. So it returns back to the calling function and increments to the next index value to check. The result is that you get a address value for the IN endpoint of 0x85 and later the macro EP_ABS_IDX() calculates the index value of the endpoint and gets 11 which is tested against NUM_OF_EP_MAX which is 8.

jfischer-phytec-iot, 2020-08-20:
The NUM_OF_EP_MAX should be double of num_bidir_endpoints, the same for the number of s_ep_ctrl,
`static usb_ep_ctrl_data_t s_ep_ctrl[NUM_OF_EP_MAX];` There is also no need to reserve endpoint addresses. With the changes below, the sample enumerates but does not work proper. Will look at it latter.

```diff
diff --git a/drivers/usb/device/usb_dc_mcux_ehci.c b/drivers/usb/device/usb_dc_mcux_ehci.c
index 5bb3fec05b..7162f4d929 100644
--- a/drivers/usb/device/usb_dc_mcux_ehci.c
+++ b/drivers/usb/device/usb_dc_mcux_ehci.c
@@ -37,7 +37,7 @@ extern void USB_DeviceEhciIsrFunction(void *deviceHandle);
 /* Then endpoint number/index calculation */
 #define EP_ABS_IDX(ep)         (USB_EP_GET_IDX(ep) * 2 + \
                                        (USB_EP_GET_DIR(ep) >> 7))
-#define NUM_OF_EP_MAX          DT_INST_PROP(0, num_bidir_endpoints)
+#define NUM_OF_EP_MAX          DT_INST_PROP(0, num_bidir_endpoints) * 2
 
 /* The minimum value is 1 */
 #define EP_BUF_NUMOF_BLOCKS    ((NUM_OF_EP_MAX + 3) / 4)
@@ -143,6 +143,7 @@ int usb_dc_set_address(const uint8_t addr)
 
 int usb_dc_ep_check_cap(const struct usb_dc_ep_cfg_data *const cfg)
 {
+       uint8_t ep_abs_idx =  EP_ABS_IDX(cfg->ep_addr);
        uint8_t ep_idx = USB_EP_GET_IDX(cfg->ep_addr);
 
        if ((cfg->ep_type == USB_DC_EP_CONTROL) && ep_idx) {
@@ -150,23 +151,11 @@ int usb_dc_ep_check_cap(const struct usb_dc_ep_cfg_data *const cfg)
                return -1;
        }
 
-       if (ep_idx >= NUM_OF_EP_MAX) {
+       if (ep_abs_idx >= NUM_OF_EP_MAX) {
                LOG_ERR("endpoint index/address out of range");
                return -1;
        }
 
-       if (ep_idx & BIT(0)) {
-               if (USB_EP_GET_DIR(cfg->ep_addr) != USB_EP_DIR_IN) {
-                       LOG_INF("pre-selected as IN endpoint");
-                       return -1;
-               }
-       } else {
-               if (USB_EP_GET_DIR(cfg->ep_addr) != USB_EP_DIR_OUT) {
-                       LOG_INF("pre-selected as OUT endpoint");
-                       return -1;
-               }
-       }
-
        return 0;
 }
 
@@ -388,14 +377,14 @@ int usb_dc_ep_disable(const uint8_t ep)
 
 int usb_dc_ep_flush(const uint8_t ep)
 {
-       uint8_t ep_idx = USB_EP_GET_IDX(ep);
+       uint8_t ep_abs_idx = EP_ABS_IDX(ep);
 
-       if (ep_idx >= NUM_OF_EP_MAX) {
+       if (ep_abs_idx >= NUM_OF_EP_MAX) {
                LOG_ERR("Wrong endpoint index/address");
                return -EINVAL;
        }
 
-       LOG_DBG("Not implemented, ep 0x%02x, idx %u", ep_idx, ep);
+       LOG_DBG("Not implemented, ep 0x%02x, idx %u", ep_abs_idx, ep);
 
        return 0;
 }
@@ -469,7 +458,6 @@ static void update_control_stage(usb_device_callback_message_struct_t *cb_msg,
 int usb_dc_ep_read_wait(uint8_t ep, uint8_t *data, uint32_t max_data_len,
                        uint32_t *read_bytes)
 {
-       uint8_t ep_idx = USB_EP_GET_IDX(ep);
        uint8_t ep_abs_idx = EP_ABS_IDX(ep);
        uint32_t data_len;
        uint8_t *bufp = NULL;
@@ -479,7 +467,7 @@ int usb_dc_ep_read_wait(uint8_t ep, uint8_t *data, uint32_t max_data_len,
                return -EBUSY;
        }
 
-       if ((ep_idx >= NUM_OF_EP_MAX) || (USB_EP_GET_DIR(ep) != USB_EP_DIR_OUT)) {
+       if ((ep_abs_idx >= NUM_OF_EP_MAX) || (USB_EP_GET_DIR(ep) != USB_EP_DIR_OUT)) {
                LOG_ERR("Wrong endpoint index/address/direction");
                return -EINVAL;
        }

```

dleach02, 2020-08-26:
@jfischer-phytec-iot I have a PR in the NXP HAL to address the divide by 2. 

zephyrproject-rtos/hal_nxp#54

jfischer-phytec-iot, 2020-08-27:
@dleach02 Sorry for delay, I have opened #27824 and also placed explanation in the commit message.

PR #27824 (2020-09-05): drivers: usb_dc_mcux_ehci: fix endpoints index assignment
MCUX EHCI USB device controller supports a specific
number of bidirectional endpoints. Bidirectional means
that an endpoint object is represented to the outside
as an OUT and an IN Eindpoint with its own buffers
and control structures.

ep_abs_idx index refers to the corresponding control
structure, for example:

```
 EP addr | ep_idx | ep_abs_idx
-------------------------------
 0x00    | 0x00   | 0x00
 0x80    | 0x00   | 0x01
 0x01    | 0x01   | 0x02
 0x81    | 0x01   | 0x03
 ....    | ....   | ....
```

The NUM_OF_EP_MAX (and number  [... cut]
closed 2020-09-05 by MaureenHelm