zephyrproject-rtos/zephyr #27576: sample.drivers.sample.drivers.peci failed to run

Zhaoningx, 2020-08-14:
Describe the bug
building tests/kernel/device/ on mec15xxevb_assy6853 board, it would report a build error.

To Reproduce
Steps to reproduce the behavior:

    west build -b mec15xxevb_assy6853 -p auto samples/drivers/
    ~/software/ci/tools/usbgpio_ftdi/pre_script.sh
    ~/software/ci/tools/usbgpio_ftdi/post_script.sh
   west flash

    see error

Logs and console output
*** Booting Zephyr OS build zephyr-v2.3.0-1774-g34859d0d3e40  ***
PECI sample test
00
00
00
00
00

peci_resp 0
00
00
00
00
00

peci_resp 0
00
00
00
00
00

peci_resp 0
Maximum temperature: 0
Start thread...
Get temp failed -5
Get temp failed -5
Get temp failed -5
Get temp failed -5

Environment (please complete the following information):

    OS: Fedora28
    Toolchain: Zephyr-sdk-0.11.4
    Commit ID: 34859d0d3e

albertofloyd, 2020-08-14:
Not supported in EVB, only in modular MEC mated to Intel RVP.

Zhaoningx, 2020-08-17:
> Not supported in EVB (mec15xxevb_assy6853 ) only in modular MEC (mec1501modular_assy6885) mated to Intel RVP.

OK, got it, close the PR later.

Zhaoningx, 2020-08-17:
Hi @albertofloyd , so, you mean to add a platform_exclude tag at sample.yaml to exclude the platform mec15xxevb_assy6853?

albertofloyd, 2020-08-17:
Yes.
Note that theoretically EVB can be connected to Intel RVP as well, but HW setup is not documented, hence qualifying as unsupported.

Zhaoningx, 2020-08-18:
OK, so I'll close this issue later.

PR #27632 (2020-08-18): samples: derives/peci can't run at EVB board
Add a tag at sample.yaml to exclude the platform mec15xxevb_assy6853
because of unsupported. Fixes #27576.
closed 2020-08-18 by nashif