zephyrproject-rtos/zephyr #27580: west install error

AbramQuerashi, 2020-08-14:
**Describe the bug**
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?

I tried to install zephyr and while installing west i got the below error.

pip3 install --user -U west
WARNING: The directory '/home/biju/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
WARNING: pip is configured with locations that require TLS/SSL, however the ssl module in Python is not available.
WARNING: The directory '/home/biju/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Collecting west

**To Reproduce**
Steps to reproduce the behavior:
pip3 install --user -U west

mbolivar-nordic, 2020-08-17:
This appears to be a pip misconfiguration. What operating system are you using?

mbolivar-nordic, 2020-08-17:
Closing this one as a duplicate of https://github.com/zephyrproject-rtos/zephyr/issues/27580, which is filed in the correct repository and has more information.

closed 2020-08-17 by mbolivar-nordic