zephyrproject-rtos/zephyr #27583: sanitycheck does not detect a linker error

chrta, 2020-08-14:
**Describe the bug**

Sanitycheck does not detect a linker error, e.g. if the SRAM overflows.

**To Reproduce**

On the current master revision 942146ff5b361990335feb90c8f89f19419335af

Steps to reproduce the behavior:
```
$ sanitycheck -p efm32hg_slstk3400a -s tests/kernel/common/kernel.common

Renaming output directory to /home/user/projects/zephyrproject/zephyr/sanity-out.4
INFO    - JOBS: 8
INFO    - Building initial testcase list...
INFO    - 1 test configurations selected, 0 configurations discarded due to filters.
INFO    - Adding tasks to the queue...
INFO    - Total complete:    1/   1  100%  skipped:    1, failed:    0
INFO    - 0 of 0 tests passed (0.00%), 0 failed, 1 skipped with 0 warnings in 5.79 seconds
INFO    - In total 27 test cases were executed on 1 out of total 274 platforms (0.36%)
INFO    - 0 tests executed on platforms, 0 tests were only built.
```

And directly checking the return code of sanitycheck:
```
$ echo $?
0
```

But:
```
$  cat sanity-out/efm32hg_slstk3400a/tests/kernel/common/kernel.common/build.log

.....

 SRAM:        8312 B         8 KB    101.46%
        IDT_LIST:          40 B         2 KB      1.95/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/../../../../arm-zephyr-eabi/bin/ld: zephyr_prebuilt.elf section `noinit' will not fit in region `SRAM'
/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/../../../../arm-zephyr-eabi/bin/ld: section .intList VMA [0000000020002000,0000000020002027] overlaps section noinit VMA [0000000020000e30,0000000020002077]
/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/../../../../arm-zephyr-eabi/bin/ld: region `SRAM' overflowed by 120 bytes
collect2: error: ld returned 1 exit status
%
zephyr/CMakeFiles/zephyr_prebuilt.dir/build.make:117: recipe for target 'zephyr/zephyr_prebuilt.elf' failed
make[2]: *** [zephyr/zephyr_prebuilt.elf] Error 1
CMakeFiles/Makefile2:2194: recipe for target 'zephyr/CMakeFiles/zephyr_prebuilt.dir/all' failed
make[1]: *** [zephyr/CMakeFiles/zephyr_prebuilt.dir/all] Error 2
Makefile:102: recipe for target 'all' failed
make: *** [all] Error 2
```

**Expected behavior**

Sanitycheck should detect the error and show this as failed.

chrta, 2020-08-14:
According to  `git bisect` the offending commit seems to be 4522e10d6513cb1872462181c41ef93e73cccd34

nashif, 2020-08-25:
As noted above, this is intentional. Code changes all the time and on a constant basis and most of the CI tests are about building and not about flashing on a certain device and running it. Having to chase all of those failures and fix tests to either be skipped is a significant overhead and would exclude platforms that would actually be able to build and run the tests. 

The tests are marked as skipped with a reason and those can be fixed if the sample/test in question has to run on the target device.

nashif, 2020-08-25:
lowering priority and leave some time for discussion.

chrta, 2020-08-26:
OK, i would also be fine to remove the bug label.

@nashif Would it make sense to add a new commandline option that enables the behaviour that an error code ( != 0) is returned in case of skipped tests because of a linker error?

nashif, 2020-08-26:
> @nashif Would it make sense to add a new commandline option that enables the behaviour that an error code ( != 0) is returned in case of skipped tests because of a linker error?

yeah, one more option to the script won't hurt, we already have a few of them :-D

PR #28372 (2020-12-01): Various sanitycheck fixes and improvements..
use python multiprocessing instead of a thread pool, resolves
various issues.

- [x] Device testing support with multiprocessing

Fixes #28040
Fixes #26949
Fixes #24652
Fixes #30099
Fixes #27583
commit f68146f7 (2020-12-01): sanitycheck: add option to treat overflows as errors

Add a new option that will enable treating RAM/SRAM overflows as errors
instead of skipping them, which is the default.

Fixes #27583
closed 2020-12-01 by nashif