zephyrproject-rtos/zephyr #27597: build system fails to propagate devicetree change to Kconfig

pabigot, 2020-08-15:
The canonical source for the console device is the `zephyr,console` chosen from devicetree.  On a clean build the Kconfig symbol `CONFIG_UART_CONSOLE_ON_DEV_NAME`, which is what actually selects the console device, defaults to that property:

```
tirzah[87]$ cd samples/hello_world
tirzah[88]$ rm -rf build && west build -b nrf52840dk_nrf52840
tirzah[89]$ grep zephyr,console build/zephyr/zephyr.dts 
		zephyr,console = &uart0;
tirzah[90]$ grep CONSOLE_ON_ build/zephyr/.config 
CONFIG_UART_CONSOLE_ON_DEV_NAME="UART_0"
```

Now edit `boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts` to change `zephyr,console = &uart1;`.

```
tirzah[127]$ west build
tirzah[128]$ grep zephyr,console build/zephyr/zephyr.dts 
		zephyr,console = &uart1;
tirzah[129]$ grep CONSOLE_ON_ build/zephyr/.config 
CONFIG_UART_CONSOLE_ON_DEV_NAME="UART_0"
```

The devicetree assignment changed, but the Kconfig that was set from devicetree originally was not updated.

`west -p auto` doesn't trigger a proper rebuild either.

Since developers may be unaware that `zephyr,console` doesn't *really* select the console device much time can be wasted figuring out where the output's actually going.

mbolivar-nordic, 2020-08-17:
This is not a bug. You are facing https://docs.zephyrproject.org/latest/guides/kconfig/tips.html#stuck-symbols-in-menuconfig-and-guiconfig in another form.

mbolivar-nordic, 2020-08-17:
> `west -p auto` doesn't trigger a proper rebuild either.

The auto heuristic has no special knowledge not available to kconfig.

closed 2020-08-17 by mbolivar-nordic

pabigot, 2020-08-17:
Thanks, I see this now.  The documented behavior of `zephyr,chosen` makes this not a bug.  It does not select the chosen UART device as people might expect; it provides the default value for a Kconfig variable that selects the chosen UART device.