zephyrproject-rtos/zephyr #27599: bluetooth shell deadlock on USB shell UART

pabigot, 2020-08-15:
`tests/bluetooth/shell` on `nrf52840dk_nrf52840` builds, and when I type `bt init` the subsystem initializes.

If I use [TBD] in [TBD] to switch to using a USB console that's also the shell device, the command line `bt_init` never returns.

Debugging shows that `bluetooth/shell/bt.c` successfully invokes `bt_enable(bt_ready)` and returns, but `bt_ready()` hangs in the `shell_print` marking the completion of initialization.  Additional diagnoses shows that `shell_pend_on_txdone()` is never receiving the signal to release it.

This behavior is reproduced both without and with #27025.

carlescufi, 2020-09-09:
@pabigot can you test it over UART as well? I suspect that this might be related to #25672

pabigot, 2020-09-09:
IIRC I did test it that way since it's the default configuration, and it worked fine.  It possibly is related; the CDC ACM UART appears to use the system work queue to complete transmission, while a real UART would do it though interrupts.  If `bt_enable()` is also using it, and trying to print from it, that could be a problem.

emob-nordic, 2020-09-14:
It is a problem but I couldn't figure out how to overcome that. Similar situation happens with mcumgr #27727.

pabigot, 2020-09-14:
Results of some investigation: `bt_enable()` invokes `bt_init()` from the system work queue, which blocks until the initialization completes.  While it's blocked work items have been queued for `cdc_acm_irq_callback_work_handler` that would have to be processed to unblock it.

This can be resolved by adding a dedicated work queue in `subsys/usb/class/cdc_acm.c`, but that doesn't help because `usb_transfer_work()` is also using the system work queue and can't make progress until `bt_init()` gets out of the way.

This can be resolved by adding a dedicated work queue in `subsys/usb/usb_transfer.c`.

Solutions are in the updated https://github.com/pabigot/zephyr/commits/issue/27599 (which also includes #28369 as an alternative to #28339).  However putting a clone of the system work queue in every module that wants to use a work queue is not sustainable.

I think the use of the system work queue in USB and Bluetooth both need to be reviewed.

It may also be that the shell needs to document that `shell_fprintf()` should not be used from anything that might be invoked by a work queue.

FYI @jfischer-phytec-iot @joerchan

pabigot, 2020-09-14:
It's actually sufficient to give the USB infrastructure its own work queue, and let `usb_transfer` and `cdc_acm` share it.  That might be a viable solution.

joerchan, 2020-09-15:
> I think the use of the system work queue in USB and Bluetooth both need to be reviewed.

Using a bluetooth work-queue instead of the system workqueue has been suggested other places, for example to free up the availability of the system-workqueue.

However this issue seems to be that the API of shell over CDC ACM cannot be called from the system-workqueue without the risking a deadlock. So solving this problem in the USB CDC ACM driver seems like the correct fix here.

Adding another workqueue could be expensive it terms of RAM though, as the work done in the callbacks could mean you need a large system workqueue stack.
Is it a possibility that the callback work is still done in the system workqueue, but the tx_work is done in another workqueue or possibly a TX thread?

pabigot, 2020-09-15:
> However this issue seems to be that the API of shell over CDC ACM cannot be called from the system-workqueue without the 
 risking a deadlock. So solving this problem in the USB CDC ACM driver seems like the correct fix here.

It's not isolated to shell; any blocking UART transmission over CDC ACM will suspend whatever thread is animating the UART flow.  The impact is not localized to tasks within the CDC ACM driver since it depends on other parts of USB to get and put data, so the fix can't be done within CDC-ACM itself.

Dunno who wants to take this on, but if I were to do it I'd have an Kconfig-controlled selection of the work queue used by the USB subsystem, either system or its own, and have the USB-specific one default selected when CDC-ACM is enabled. Too much functionality depends on UART being able to make progress regardless of which thread is waiting for it.  I can submit that as a fix if nobody else has a different approach in progress.

At this time I don't recommend changes to Bluetooth, just taking into account that other system work queue users can block it and considering future changes.

PR #28408 (2020-09-16): usb: add support for dedicated work queue
Add an optional work queue to mitigate deadlocks when the USB infrastructure (servicing CDC-ACM UART) is blocked by non-USB tasks on the system work queue that invoke UART operations assuming these can progress without putting the task to sleep.

Note per https://github.com/zephyrproject-rtos/zephyr/issues/27599#issuecomment-692214683 that fixes isolated to the CDC-ACM driver are not sufficient, as usb-transfer must also be able to make progress.

Fixes #27599
closed 2020-09-16 by MaureenHelm