zephyrproject-rtos/zephyr #27611: Power Control using PPP connection (SARA U201)

mtahirbutt, 2020-08-17:
**Is your enhancement proposal related to a problem? Please describe.**
A clear and concise description of what the problem is.

I want to reduce power usage of modem when there is no data in the main application. But net_if_up() and net_if_down are not successful in achieving using PPP although offload works for this purpose.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Either net_if_up and net_if_down should directly control it. or power reset to the modem should reestablish new PPP connection.
**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

I have tried a couple for things in this regard, 
1- tried to see if I could issue AT commands by resetting the modem. But it seems like PPP uses UART whereas in case of offload, modem iface uses the UART. so GSM PPP driver does not receive any UART data after modem reset.
Right now I am trying GSM muxing to see if I could see better results.

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

Some ideas are required in this enhancement.

endian-benjamin, 2020-08-17:
I don't think doing it in `net_if_down` is feasible, as explained by this issue that I just created: https://github.com/zephyrproject-rtos/zephyr/issues/27622

kartben, 2026-10-30:
Closing as obsolete: the gsm_ppp driver this refers to has been replaced by the modem_cellular driver, which handles modem power management differently. If you need this with modem_cellular on current main, please open a new issue.

closed 2026-10-30 by kartben