zephyrproject-rtos/zephyr #27613: CONFIG_ASSERT not working on nrf5340dk_nrf5340_cpunet in hci_rpmsg sample

alexsven, 2020-08-17:
**Describe the bug**
When running the hci_rpmsg + central_hr/peripheral_hr on two nrf5340DK, if I set CONFIG_ASSERT=y in prj.conf in hci_rpmsg it doesn't start. I get *** Booting Zephyr OS build zephyr-v2.3.0-1883-g4bd247af700a  *** but nothing else. This is not an issue when compiling for nrf5340pdk, however.

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: gnuarmemb
 - Commit SHA: 4bd247af700a

alexsven, 2020-08-24:
> I wonder @alexsven if you could bisect and find an offending commit.
> Of course it has to be after the support for DK was introduced..

Tested with #26110 (first PR with support for the DK) and it didn't work there

ioannisg, 2020-08-24:
Thanks @alexsven , I am taking a look.

PR #27742 (2020-08-25): Boards nrf5340dk fix net num mpu regs
When we introduced the support for nRF5340 DK, we did not add a separate .dtsi header to reflect that the number of MPU regions are different compared to the PDK, _also_ for the NETWORK MCU. This PR fixes this.

Fixes #27613
closed 2020-08-25 by ioannisg