zephyrproject-rtos/zephyr #27622: Power management for modems using PPP

endian-benjamin, 2020-08-17:
Currently the GSM PPP driver lacks power management, which is a real dealbreaker for many use cases. Indeed it basically makes Zephyr infeasible for cellular devices that wants to run on battery power.

Note that power management does not belong in `net_if_down() / ppp_stop()`
because a modem may present more than a network interface - indeed that's why the
PPP muxing functionality was added in the first place - SMS, GPS, network
wallclock etc have nothing to do with the networking stack and should be able to
exist even if the network interface is down.

For this reason, we should separate power management from the networking stack completely. But `gsm_ppp.c` defines a device for the modem itself, so that might be the only logical place for a PM API at the moment.

### Suggestions
 - generic device pm API in `gsm_ppp.c` using `DEVICE_DEFINE`
 - when using `CONFIG_DEVICE_POWER_MANAGEMENT`, setting up PPP (`ATD*99#`) and enabling GSM muxing should not be done in `gsm_init`, it should be done when modem is POWERED UP
 - all relevant setup functions needs to be ensured to be reentrant
 - powering down the modem should bring down the network interface first
 - `net_if_up() / ppp_start()` should ensure that the modem is powered up

Note: There's currently a lot of `modem_*` files that tries very hard to handle all modems in the same way but a lot of PM related stuff will be device dependent. How to accommodate this
cleanly in `gsm_ppp`? Weak linkage? A `<modem>_pm.c` file for each modem?
`#ifdef`s????

endian-benjamin, 2020-08-17:
@mike-scott @weinholtendian @jukkar

mtahirbutt, 2020-08-17:
the problem can be disected into two subtasks, how to handle UART to get data from modem via PPP, as PPP reserves UART device, on the contrary offload does not work like this. In the second step, connection manager has to take care connection is re-established correctly when we reconnect PPP.

jukkar, 2020-08-17:
> A `<modem>_pm.c` file for each modem?

If we are talking about modems that use generic GSM PPP, then there is basically only one modem driver gsm_ppp.c. All the other modem related drivers are for offloading use and they do not use PPP. I would leave the offloading modems out of this right now in order to simplify things.

endian-benjamin, 2020-08-18:
@jukkar Once you've enabled GSM PPP then they should ideally behave very similar. But first there's a power on sequence. And after PPP is stopped, there's a powerdown sequence. For instance, UBlox strongly recommends sending `AT+CPWROFF`  and waiting for `OK` before powering off - not sure how Simcom wants it to be done but it 100% will differ between modems.

But! We could, in `gsm_ppp.c`, provide generic, naive power functions, something like:

```
void __weak modem_set_power_state(struct device *dev, uint32_t state)
{
    gpio_pin_write(DT_MDM_PWR_GPIO, !!state);
}
```

which could be overriden either by a driver or a board file. I'm not sure what the policy on weak linkage is though- some people hate it due to poor interactions with `-flto` (under some versions of GCC).

jukkar, 2020-08-18:
> Once you've enabled GSM PPP then they should ideally behave very similar. But first there's a power on sequence. And after PPP is stopped, there's a powerdown sequence

Yep, very true. I misunderstood what you meant in earlier comment.

> void __weak modem_set_power_state(struct device *dev, uint32_t state)

Weak function is of course a possibility, but it restricts the system to have only one kind of modems. If we have a proper API, we could at least in theory to have multiple different brand modems connected to zephyr. This might be a far fetched use case but why we would restrict ourselves here.

endian-benjamin, 2020-08-18:
I believe that assumption is already hard coded into `gsm_ppp.c` :)

endian-benjamin, 2020-09-17:
https://github.com/zephyrproject-rtos/zephyr/pull/28475

jukkar, 2020-11-17:
Closing this as minimal PM to GSM modem is already merged.

closed 2020-11-17 by jukkar