zephyrproject-rtos/zephyr #27628: stm32: i2c bus failure when using USB

bpbradley, 2020-08-17:
I am creating this issue as a starting point. When I get some time I will need to gather more information and work towards finding a minimal way to repro so that others can help guide me towards a solution. But for now this issue is a bit of a blocker for my application as it renders I2C pretty much unusable on stm32.

**Describe the bug**
I have an application (targeting stm32f411e) which uses USB as a console device in the CDC mode of operation, as well as USB as a mass storage device. On their own, these work perfectly.

The application also has a couple of (out-of-tree for now) I2C sensors, built into zephyr with the sensor API. I eventually intend to upstream these, but I do not believe there is anything wrong with the specific implementation of these sensors. 

I am polling these two sensors once per second in sequence. On their own, these work perfectly fine. If I power the device with a standalone power supply, and don't perform any USB communication, the application operates as expected for days on end.

However, when I actually run the application with both USB and I2C active, issues (eventually) occur. If I open the console (writing logs), or access the mass storage device periodically, for example, then eventually both USB and I2C hang. Other threads which do not rely on those subsystems continue to operate normally.

I have captured a view of SCL under normal conditions here

[image]

And the moment the hang up begins, this is the output

[image]

At this point, no activity occurs on SCL.

After some discussion with other users on Slack, I eventually tested operation with `CONFIG_I2C_STM32_INTERRUPT=n`

This stopped the problem from occuring (in the same way). Normally, the hang up would occur after 1-5 minutes. Now, it operates fine for 45 minutes to an hour, but once again crashes. However, it no longer crashes the USB, only I2C crashes. Now, the thread with sensor operations no longer blocks, and the sensor fetch function returns -EIO. 

Enabling I2C_LOG_LEVEL_DBG shows there is something going on in the msg_end function as I continuously get 

`<dbg> i2c_ll_stm32_v1.msg_end: stm32_i2c_msg_write: NACK`

I tried recovering the I2C bus with `i2c recover` from the shell, but this does not appear to be implemented (perhaps I can figure out a recovery method and add the api function for i2c_recover?)

That is as far as I have been able to debug so far, but will have more time in the coming days to look into this.

**To Reproduce**
I do not yet have steps to easily reproduce this without writing some minimal code. I will work on this later in the week.

**Expected behavior**
I2C should work as normal, even if USB is active. There should be a means of recovery if the bus does crash. `CONFIG_I2C_STM32_INTERRUPT` should not cause I2C and USB to crash in the event of a bus failure.

**Impact**
Showstopper for my application.

**Environment (please complete the following information):**
 - Windows 10 OS
 - gnuarmemb toolchain
 - tested on latest master

**Additional context**
This problem does not occur on nordic hardware, with all else held constant (same wiring/sensors, etc).

lowlander, 2020-08-18:
There still seems to be some places in the I2C driver that can cause an endless IRQ stream when something goes wrong on the bus (EMC spikes). I have seen this today with my hardware, it just endlessly generates IRQ's but never gets ready, since the load on the system gets so high that the timeout code never gets executed and my hardware watchdog resets the system.

The fixes it the driver that reset the bus when something goes wrong simply never get executed because of the high IRQ load, so there is certainly still a bug in the driver. 

When using the non-irq driver the hang also happens, but since it doesn't endlessly generate IRQ's the rest of the system keeps functioning. I added the bus-reset code to the non-irq error path and that seems to make things better for me. 

@bpbradley can you see in the debugger what the I2C driver is doing when it hangs? Is it also endlessly handling IRQ's ?

bpbradley, 2020-08-18:
> There still seems to be some places in the I2C driver that can cause an endless IRQ stream when something goes wrong on the bus (EMC spikes). I have seen this today with my hardware, it just [...]

Thanks for the comment, and I appreciate the help you gave in Slack. I will try to test with a debugger attached this afternoon if I can. Should I test with STM32_INTERRUPT enabled or disabled? I assume enabled.

Your theory makes sense though. I think I am realizing my specific problem is being caused by some emc issue as you say (my prototype is breadboarded right now), which is why creating a repro procedure with different hardware might be tricky, but it is the inability to recover which is problematic. The same exact wiring on nordic hardware does not present this issue.

If we have a pinctrl api which allows me to re-purpose sda/scl at runtime to implement the i2c bus reset sequence in the bus_recover api, then I think it will at least allow me to recover in the non-interrupt mode of operation. But the total system failure associated with the interrupt mode of operation is worrisome.

lowlander, 2020-08-18:
@bpbradley I made a PR #27653 that resets the I2C hardware on error when using the non-irq setting. Can you give this a try an see if your software recovers from the error?

bpbradley, 2020-08-18:
> @bpbradley I made a PR #27653 that resets the I2C hardware on error when using the non-irq setting. Can you give this a try an see if your software recovers from the error?

Testing now. Will probably know if it worked within an hour.

bpbradley, 2020-08-18:
@lowlander For some reason it took a long time to reproduce this time. But it finally did, but still didn't recover properly. It did result in different behavior still though. I now get the log message `<dbg> i2c_ll_stm32_v1.stm32_i2c_generate_start_condition: stm32_i2c_generate_start_condition: START while STOP active!
`
Again in this case, manually asserting SDA fixed the issue and allowed it to properly recover.

closed 2020-11-02 by github-actions[bot]

erwango, 2020-11-02:
@bpbradley don't hesistate to re-open if issues shows up again.