zephyrproject-rtos/zephyr #27636: sensor: shell float output broken w/ CONFIG_NEWLIB_LIBC=y

drandreas, 2020-08-18:
**Describe the bug**
Executing `sensor get TEMP1` on FRDM_K64 returns:
```
uart:~$ 
channel idx=12 die_temp = %.6f
channel idx=30 voltage = %.6f
uart:~$
```

I've traced the printf call from `drivers/sensor/sensor_shell.c` to `subsys/shell/shell_fprintf.c` and expect the error to be related to the following code snippet:
```
#if !defined(CONFIG_NEWLIB_LIBC) && !defined(CONFIG_ARCH_POSIX)
  (void)z_prf(out_func, (void *)sh_fprintf, (char *)fmt, args);
#else
  z_vprintk(out_func, (void *)sh_fprintf, fmt, args);
#endif
```

I was however not able to resolve the issue.

**To Reproduce**
Steps to reproduce the behavior:
```
$ echo CONFIG_ADC=y >> zephyr/samples/sensor/sensor_shell/prj.conf
$ echo CONFIG_TEMP_KINETIS=y >> zephyr/samples/sensor/sensor_shell/prj.conf
$ echo CONFIG_NEWLIB_LIBC=y >> zephyr/samples/sensor/sensor_shell/prj.conf
$ west build -p -b frdm_k64f zephyr/samples/sensor/sensor_shell
$ west flash
$ miniterm.py --raw /dev/ttyACM0 115200
$ sensor get TEMP1
```

**Expected behavior**
```
uart:~$ sensor get TEMP1
channel idx=12 die_temp =  33.710000
channel idx=30 voltage =   3.293000
uart:~$
```

**Impact**
showstopper: `sensor get` can not be used at all.

**Environment (please complete the following information):**
 - Ubuntu 18.04
 - zephyr-sdk-0.11.3
 - eb9a1843248c35ebda66d8857264f92000eefcc9

pabigot, 2020-08-18:
Thanks for reporting this; I'd discovered it some time back while supporting a slack question, but didn't follow up.

Should be resolved by #27655, which also explains why this was done.

drandreas, 2020-08-18:
That was a quick fix. Thank you very much.

PR #27655 (2020-09-03): shell: support printing floating point values with newlib
The shell requires a `*printf` function that produces its output by calling a function like `putchar(3c)` to output one character at a time.  This eliminates dynamic or static allocation of buffers large enough to hold arbitrary messages.

When `CONFIG_MINIMAL_LIBC=y` a custom implementation `z_prf()` is available which is capable of producing float values.  It's used to implement `s*printf()` in minimal libc, and is part of the minimal libc library.

When `CONFIG_NEWLIB_LIBC=y` the minimal libc library is not available, so `z_prf()` is [... cut]
closed 2020-09-03 by carlescufi