zephyrproject-rtos/zephyr #27639: [Coverity CID :212425] Out-of-bounds access in drivers/ethernet/eth_mcux.c

zephyrbot, 2020-08-18:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/66bd06a7d1f9e4682faafbc551046af695fa1060/drivers/ethernet/eth_mcux.c#L1030

Category: Memory - corruptions
Function: `eth_mcux_init`
Component: Drivers
CID: [212425](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=212425)

Details:

```
1024     #endif
1025    
1026         context->phy_state = eth_mcux_phy_state_initial;
1027    
1028         sys_clock = CLOCK_GetFreq(kCLOCK_CoreSysClk);
1029    
>>>     CID 212425:  Memory - corruptions  (OVERRUN)
>>>     Overrunning struct type enet_config_t of 36 bytes by passing it to a function which accesses it at byte offset 39.
1030         ENET_GetDefaultConfig(&enet_config);
1031         enet_config.interrupt |= kENET_RxFrameInterrupt;
1032         enet_config.interrupt |= kENET_TxFrameInterrupt;
1033     #ifndef CONFIG_ETH_MCUX_NO_PHY_SMI
1034         enet_config.interrupt |= kENET_MiiInterrupt;
1035     #endif
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

carlescufi, 2020-09-08:
@agansari could you please take a look at this issue?

agansari, 2020-09-09:
@carlescufi sorry i can't find the CID on the coverity site.
It's a non issue, depends on how the coverage software interprets the macros used in definition and function.

carlescufi, 2020-09-15:
Marked as fixed in Coverity

closed 2020-09-15 by carlescufi