zephyrproject-rtos/zephyr #27647: [Coverity CID :212429] Negative array index write in tests/subsys/fs/fs_api/src/test_fs_dir_file.c

zephyrbot, 2020-08-18:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/66bd06a7d1f9e4682faafbc551046af695fa1060/tests/subsys/fs/fs_api/src/test_fs_dir_file.c#L478

Category: Memory - corruptions
Function: `test_file_read`
Component: Tests
CID: [212429](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=212429)

Details:

```
472         brw = fs_read(&filep, NULL, sz);
473         zassert_false(brw >= 0, "Read data to a invalid buffer");
474    
475         brw = fs_read(&filep, read_buff, sz);
476         zassert_true(brw >= 0, "Fail to read file");
477    
>>>     CID 212429:  Memory - corruptions  (NEGATIVE_RETURNS)
>>>     Using variable "brw" as an index to array "read_buff".
478         read_buff[brw] = 0;
479         TC_PRINT("Data read:\"%s\"\n\n", read_buff);
480    
481         zassert_true(strcmp(test_str, read_buff) == 0,
482                 "Error - Data read does not match data written");
483    
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

de-nordic, 2020-08-19:
Not an issue. The line 476 should root out all negative values by failing test, before execution reaches problematic line 478.

wentongwu, 2020-09-09:
agree, negative values have already been filtered out by line 476` zassert_true(brw >= 0, "Fail to read file");`

de-nordic, 2020-09-09:
Marked as False Positive with no further action set.

closed 2020-09-09 by de-nordic