zephyrproject-rtos/zephyr #27651: IPv4 autoconfiguration implementation is in contradiciton to RFC3927

AntoineZen, 2020-08-18:
**Describe the bug**
IPv4LL also called Auto-IP, "Local Link" or "IPv4 autoconfiguration" in Zephyr is enabled by setting `CONFIG_NET_IPV4_AUTO`. [The documeation](https://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_NET_IPV4_AUTO.html#cmdoption-arg-config-net-ipv4-auto) states that the implementation is base on RFC3927.

When `CONFIG_NET_IPV4_AUTO` is used `CONFIG_NET_IF_UNICAST_IPV4_ADDR_COUNT` is set to 2 instead of 1 (see [here](https://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_NET_IF_UNICAST_IPV4_ADDR_COUNT.html?highlight=config_net_if_unicast_ipv4_addr_count#defaults) ). When DHCP is used, this allows the interface to have BOTH an DHCP assigned  and local-link IPv4 addresses. This can be seen using the `net iface` shell command:

```
zephyr$ net iface 

Interface 0x20259220 (Ethernet) [1]
===================================
Link addr : D4:66:A8:10:00:B8
MTU       : 1500
Ethernet capabilities supported:
	TX checksum offload
	RX checksum offload
	Virtual LAN
	Auto negotiation
	10 Mbits
	100 Mbits
IPv4 unicast addresses (max 2):
	169.254.148.77 autoconf preferred infinite
	10.0.0.10 DHCP preferred
IPv4 multicast addresses (max 1):
	<none>
IPv4 gateway : 0.0.0.0
IPv4 netmask : 255.255.255.0
DHCPv4 lease time : 519
DHCPv4 renew time : 259
DHCPv4 server     : 10.0.0.1
DHCPv4 requested  : 10.0.0.10
DHCPv4 state      : bound
DHCPv4 attempts   : 1
```

Howerver, [RFC3927 section 1.9] (https://tools.ietf.org/html/rfc3927#section-1.9) states:

> When an operable routable address is available on an interface, the host SHOULD NOT also assign an IPv4 Link-Local address  on that interface.

Here, Zephyr is not following the specification, as the DHCP assigned is a "routable addess". Also, Windows and GNU/Linux uses Auto-IP only as a fall-back when DHCP process does not success.

I've try to force `CONFIG_NET_IF_UNICAST_IPV4_ADDR_COUNT` to 1, but then autoconf and DHCP start to race and if autoconf succeed before DHCP, then DHCP cannot assign the revived address and re-initialize the discovery/request process.

**To Reproduce**
1) Prepare the "Sample DHCPv4 client application" sample as described [here](https://docs.zephyrproject.org/latest/samples/net/ipv4_autoconf/README.html).
2) Modify `samples/net/dhcpv4_client/prj.conf` and add `CONFIG_NET_IPV4_AUTO=y`.
3) Optionnaly, add `CONFIG_NET_IF_UNICAST_IPV4_ADDR_COUNT=1`

**Expected behavior**
I see two possible behavior: 
1) DHCP have precedence over autoconf. If `CONFIG_NET_IPV4_AUTO=y` and `CONFIG_NET_DHCPV4=y`  and  `CONFIG_NET_IF_UNICAST_IPV4_ADDR_COUNT=1` are configured, I expect having Auto-IP that kick-in after some DHCP timeout. If later-on DHCP succeed, the Auto-ip Assinged IP is dropped. This will be "Auto-IP has DHCP fall-back" as on may desktop OS.
2) Having API to take control of autoconf or example `net_ipv4_autoconf_start()` , `net_ipv4_autoconf_stop()`) in the same fashion as the DHCP client. As now, the Auto-IP "daemon" kick-in when an interface is up automatically as soon `CONFIG_NET_IPV4_AUTO` is enabled.

**Impact**
In our application we don't want an interface to have two IPv4 address. With the current implementation this does not work because of the race between DHCP and Auto-IP. This is also in contraction with the specification (RFC3927).

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain : zephyr-sdk-0.11.1
 - Zephry: Release 2.2.

AntoineZen, 2020-08-19:
For instance, LwIP has the following option regading DHCP and Auto-IP co-ooperation:

```
/**
 * LWIP_DHCP_AUTOIP_COOP==1: Allow DHCP and AUTOIP to be both enabled on
 * the same interface at the same time.
 */
#ifndef LWIP_DHCP_AUTOIP_COOP
#define LWIP_DHCP_AUTOIP_COOP           1
#endif

/**
 * LWIP_DHCP_AUTOIP_COOP_TRIES: Set to the number of DHCP DISCOVER probes
 * that should be sent before falling back on AUTOIP. This can be set
 * as low as 1 to get an AutoIP address very quickly, but you should
 * be prepared to handle a changing IP address when DHCP overrides
 * AutoIP.
 */
#ifndef LWIP_DHCP_AUTOIP_COOP_TRIES
#define LWIP_DHCP_AUTOIP_COOP_TRIES     5
#endif
```

Having such a Kconfig configuration option would be nice! What do you think ?
Maybe just enable this co-operation if an user select `CONFIG_NET_DHCPV4` and `CONFIG_NET_IPV4_AUTO` at the same time ?

I would like to have any feed-back from the community or code-owner, then I can start a PR my-self.

jukkar, 2020-08-19:
> Howerver, [RFC3927 section 1.9] (https://tools.ietf.org/html/rfc3927#section-1.9) states:
>
>
> Here, Zephyr is not following the specification, as the DHCP assigned is a "routable addess". Also, [...]

As you noted, the spec says "... the host SHOULD NOT ..." so this is not strictly disallowed although it can cause issues later. The RFC has some conditions in chapter 1.9 that we should implement, patches are welcome here. Because of the "SHOULD NOT" wording, I am lowering the priority.

Your proposal looks sane, if DHCPv4 does not succeed, we could fall back to auto IP. Note that if DHCPv4 fails, and we start auto IP, and then if DHCPv4 comes back on, we have a similar issue on our hands. At that point the auto IP should be disabled until the DHCPv4 fails again. The bigger issue is that if the application is using the auto IP address at that point, we just cannot remove that address immediately.

cperera-aud, 2022-08-29:
Hi @jukkar , We would like understand a bit more as to your statement regarding:

> The bigger issue is that if the application is using the auto IP address at that point, we just cannot remove that address immediately

Does this still stand in v3.1? Also to clarify if the application starts with an auto IP address and then gets a DHCP address, is the new address that cannot be removed immediately? Is there a delay involved in doing that? 

Also, is there a mechanism to delay auto IP so that DHCP can win the race? 

Hi @AntoineZen , did you end up committing any changes? We'd be happy to integrate them.

jukkar, 2022-08-29:
What I meant that if the application is using auto IP address (like 169.254.148.77) to communicate to peer and then the device gets DHCP address (like 10.0.0.10), what should we do with the socket that might be connected using the auto IP address? Currently this is not handled at all.

savinweera, 2022-08-31:
Would a possible solution be to have the NET_ADDR_AUTOCONF address type be overridable by DHCP, similar to NET_ADDR_OVERRIDABLE ?

```
enum net_addr_type {
    ...
    ...
    /** Manually set address which is overridable by DHCP */
    NET_ADDR_OVERRIDABLE,
} __packed;
```

Modifying code in net/ip/net_if.c
```
if (addr_type == NET_ADDR_DHCP && 
        ((cur->addr_type == NET_ADDR_OVERRIDABLE) || (cur->addr_type == NET_ADDR_AUTOCONF))) {
	ifaddr = cur;
	break;
}
```

jukkar, 2023-11-16:
@ad8sweera would you be able to send a PR for this?

cperera-aud, 2023-11-16:
Hi @jukkar  just a heads up that ad8sweera is on leave at the moment so there might be some delay getting back to you.

jukkar, 2023-11-16:
@cperera-aud No worries, I was going through old issues and wondered if possible fixes could get to PR stage.

carlescufi, 2023-11-22:
@cperera-aud could you perhaps let us know if you (in the company) have plans to address this?

carlescufi, 2023-11-22:
> Would a possible solution be to have the NET_ADDR_AUTOCONF address type be overridable by DHCP, similar to NET_ADDR_OVERRIDABLE ?

Perhaps, maybe if you have something like this working already and you post a PR so that others can take a look?

GeorgeAtKistler, 2024-01-30:
> I've try to force CONFIG_NET_IF_UNICAST_IPV4_ADDR_COUNT to 1, but then autoconf and DHCP start to race and if autoconf succeed before DHCP, then DHCP cannot assign the received address and re-initialize the discovery/request process.

Based on my investigations in https://github.com/zephyrproject-rtos/zephyr/discussions/68238 I see the following:
Precondition: DHCP server not available.
1. After approx 12s, the device gets an AutoIP
2. The DHCP client still is running (the DHCP servers get requests every few seconds (seems to be random between 5..15s) and also returns a response):
```bash
Jan 30 15:52:46 Linux-Server dhcpd[15804]: DHCPDISCOVER from 00:80:e1:33:39:38 via enp2s0
Jan 30 15:52:46 Linux-Server dhcpd[15804]: DHCPOFFER on 10.10.2.234 to 00:80:e1:33:39:38 via enp2s0
Jan 30 15:52:46 Linux-Server dhcpd[15804]: DHCPREQUEST for 10.10.2.234 (10.10.2.224) from 00:80:e1:33:39:38 via enp2s0
Jan 30 15:52:46 Linux-Server dhcpd[15804]: DHCPACK on 10.10.2.234 to 00:80:e1:33:39:38 via enp2s0

Jan 30 15:52:51 Linux-Server dhcpd[15804]: DHCPREQUEST for 10.10.2.234 (10.10.2.224) from 00:80:e1:33:39:38 via enp2s0
Jan 30 15:52:51 Linux-Server dhcpd[15804]: DHCPACK on 10.10.2.234 to 00:80:e1:33:39:38 via enp2s0

Jan 30 15:52:58 Linux-Server dhcpd[15804]: DHCPREQUEST for 10.10.2.234 (10.10.2.224) from 00:80:e1:33:39:38 via enp2s0
Jan 30 15:52:58 Linux-Server dhcpd[15804]: DHCPACK on 10.10.2.234 to 00:80:e1:33:39:38 via enp2s0

Jan 30 15:53:14 Linux-Server dhcpd[15804]: DHCPDISCOVER from 00:80:e1:33:39:38 via enp2s0
Jan 30 15:53:14 Linux-Server dhcpd[15804]: DHCPOFFER on 10.10.2.234 to 00:80:e1:33:39:38 via enp2s0
Jan 30 15:53:14 Linux-Server dhcpd[15804]: DHCPREQUEST for 10.10.2.234 (10.10.2.224) from 00:80:e1:33:39:38 via enp2s0
Jan 30 15:53:14 Linux-Server dhcpd[15804]: DHCPACK on 10.10.2.234 to 00:80:e1:33:39:38 via enp2s0

Jan 30 15:53:18 Linux-Server dhcpd[15804]: DHCPREQUEST for 10.10.2.234 (10.10.2.224) from 00:80:e1:33:39:38 via enp2s0
Jan 30 15:53:18 Linux-Server dhcpd[15804]: DHCPACK on 10.10.2.234 to 00:80:e1:33:39:38 via enp2s0

Jan 30 15:53:25 Linux-Server dhcpd[15804]: DHCPREQUEST for 10.10.2.234 (10.10.2.224) from 00:80:e1:33:39:38 via enp2s0
Jan 30 15:53:25 Linux-Server dhcpd[15804]: DHCPACK on 10.10.2.234 to 00:80:e1:33:39:38 via enp2s0

Jan 30 15:53:42 Linux-Server dhcpd[15804]: DHCPDISCOVER from 00:80:e1:33:39:38 via enp2s0
Jan 30 15:53:42 Linux-Server dhcpd[15804]: DHCPOFFER on 10.10.2.234 to 00:80:e1:33:39:38 via enp2s0
Jan 30 15:53:42 Linux-Server dhcpd[15804]: DHCPREQUEST for 10.10.2.234 (10.10.2.224) from 00:80:e1:33:39:38 via enp2s0
Jan 30 15:53:42 Linux-Server dhcpd[15804]: DHCPACK on 10.10.2.234 to 00:80:e1:33:39:38 via enp2s0

Jan 30 15:53:46 Linux-Server dhcpd[15804]: DHCPREQUEST for 10.10.2.234 (10.10.2.224) from 00:80:e1:33:39:38 via enp2s0
Jan 30 15:53:46 Linux-Server dhcpd[15804]: DHCPACK on 10.10.2.234 to 00:80:e1:33:39:38 via enp2s0

Jan 30 15:53:53 Linux-Server dhcpd[15804]: DHCPREQUEST for 10.10.2.234 (10.10.2.224) from 00:80:e1:33:39:38 via enp2s0
Jan 30 15:53:53 Linux-Server dhcpd[15804]: DHCPACK on 10.10.2.234 to 00:80:e1:33:39:38 via enp2s0
```
 How ever the device is unable to use that response.
 My guess: It wants to assign the DHCP response to slot 2 (that's what it would be if there would be 2 slots). But this silently fails.

GeorgeAtKistler, 2024-01-30:
> Your proposal looks sane, if DHCPv4 does not succeed, we could fall back to auto IP. Note that if DHCPv4 fails, and we start auto IP, and then if DHCPv4 comes back on, we have a similar issue on our [...]

In my opinion, AutoIP is a fallback feature to make a device reachable if it does not have a static IP and does not get a DHCP IP. A user should use the AutoIP to configure the device but not keep using the device with the AutoIP. This would make it _more acceptable_ to simply replace the AutoIP with the received DHCP response. But this is only my opinion and I did not check any RFC regarding this use case.

GeorgeAtKistler, 2024-01-30:
Note that @ad8sweera had a proposal in Sept. 2022 which how ever got closed: https://github.com/zephyrproject-rtos/zephyr/pull/50245

jukkar, 2024-02-09:
> A user should use the AutoIP to configure the device but not keep using the device with the AutoIP.

What do you mean by "not keep using" here? What an application is suppose to do with the AutoIP address?