zephyrproject-rtos/zephyr #27662: High Ethernet network load starves user tasks

lowlander, 2020-08-19:
When doing a ping -f to a Zephyr STM32F407 (100Mbit) based system the network stack will use 100% CPU, and since the network stack tasks are COOP prio tasks all user tasks starve (causing a watchdog reset in my case). Making the user tasks higher COOP prio is not an option because the lower prio tasks may not block higher prio tasks in my application. 

Some sort of rate limiting must be implement, or it must be possible to lower the prio of the network tasks to non-COOP prios.

jukkar, 2020-08-19:
I would say this is more like an enhancement than a bug. If we receive lot of incoming traffic, we always end up starving things. We just need a way to limit the accepted network packets in the network driver, and some way to control that. For control, we could use the config interface. For Ethernet, see `struct ethernet_config` in `include/net/ethernet.h` for details.

lowlander, 2020-08-19:
Hmm it is a denial of service attack, which is not very good advertisement for a secure IoT OS ;-) I would consider it a bug that I as developer can not decide that my other tasks are more important (without being forced to COOP-prio) than the network tasks.

armandciejak, 2020-10-01:
We have the same issue and since in our case the networking tasks cannot have higher priority than really real-time tasks we worked around the problem by changing the number of cooperative priorities `CONFIG_NUM_COOP_PRIORITIES=5` this changes the priority of the rx and tx work queues to 2 (preemptive threads).

lowlander, 2020-10-01:
>  ... we worked around the problem by changing the number of cooperative priorities `CONFIG_NUM_COOP_PRIORITIES=5` this changes the priority of the rx and tx work queues to 2 (preemptive threads).

But that also sounds like a hack, in most cases I think network is not realtime anyway, and the user should be able to select a non-cooperative prio for the network threads.  But that might need a very good look at if the tasks may be preempted or if that would cause problems.

armandciejak, 2020-10-01:
@lowlander I agree that this is a workaround, I just mentioned it so that someone who is impacted by this problem knows that it is possible to work around the problem by changing the configuration. The down side is of course the fact that this reduces the number of cooperative priorities from 16 to 5.

jukkar, 2020-10-01:
The TX / RX thread priorities can be tweaked in `subsys/net/ip/net_tc.c` (search `tx_tc2thread()` and `rx_tc2thread()` arrays). Currently these are not configurable but if anyone has a good idea how to allow easy configuration, just send a PR.

lowlander, 2020-10-02:
@jukkar AFAIK they are currently cooperative threads, the question is can they be changed to preemptable threads? With other words is the cooperative scheduling used as a mutal exclusion mechanism that will break when the TX thread for example preempts the RX thread?

jukkar, 2020-10-02:
> the question is can they be changed to preemptable threads

Atm we have only used co-operative threads so more testing would be needed to verify if they work with pre-emptive threads. We have locking in various parts of the network stack but more testing is needed to make sure they are in proper places.

carlescufi, 2023-11-22:
@lowlander perhaps you could try using this Kconfig option:
https://github.com/zephyrproject-rtos/zephyr/blob/45a5cedfd6a0d647f6976547e3868a00b1d83269/subsys/net/ip/Kconfig#L257

clamattia, 2025-01-24:
In a similar scenario, I had great success with a specific priority setup. (Note, that this patch is needed: https://github.com/zephyrproject-rtos/zephyr/pull/84442 to avoid starvation of the queues and of the direct sending).

1. Have a single `tx_q` it will send out the ping responses. Give it the lowest preemptive thread-priority in the system.
2. For application level packets configure a high packet-priority (for example `SO_PRIORITY` socket-option on the socket). Let them bypass the `tx_q` by enabling `CONFIG_TC_TX_SKIP_FOR_HIGH_PRIO`. Give the application level threads an higher thread-priority.
3. Patch the tx-allocation timeout for some protocols (request tracked in https://github.com/zephyrproject-rtos/zephyr/issues/84482) so that `rx` is not blocked by full `tx`. So the ping requests are dropped already in the `rx` context, when incoming.
4. Give `rx_q`s a higher (in my case still preemtable) priority.
5. Optionally, configure multiple `rx_q`s and hack the driver to assign the incoming packets you are interested in a higher priority (I am currently thinking about a proposal, how this could be done in a driver independent way).
6. Give driver `rx` thread the highest COOP priority available in the system.

Such a layout is called `back pressure`, I believe. It ensures, that the system remains responsive even in such hostile environments.