zephyrproject-rtos/zephyr #27672: BLE Transaction Collision

sarazalabany, 2020-08-19:
Hello, 

I am currently developing a BLE application using Zephyr 2.2.1. 
My application involves 3 bluetooth devices, one Central and two peripherals. When connection is achieved between the devices, one of the slaves sends data periodically with 100 ms period to the master and the master passes the data packets to the other slave. 

So exactly how I am achieving this is as follows, once the master goes to the receive call back function, it sends data to the next L2CAP channel where the second slave is connected to.

The problem occurs if I do not add any delay before sending the data, the connection is dropped with the second slave and nothing happens anymore. 
I have been using a sniffer to see the data sent and I have found that  there is a transaction collision that occurs as shown in the attached photo.

I would likt to mention that a delay of 1 ms solved this issue. 

Thank you idvance..
[image]
[image]

cvinayak, 2020-08-21:
Please try with this https://github.com/zephyrproject-rtos/zephyr/pull/26404

cvinayak, 2020-08-21:
If I understand from the limited information in the screenshot of the sniffer trace, the transaction collision is a valid scenario when both master and slave are initiating the PHY update procedure. The slave gets the REJECT and master proceed with its procedure to completion. 

What is the application symptoms you are observing?

sarazalabany, 2020-08-21:
@cvinayak When the master receives the first packet from the first slave,  the master goes into the receive callback function and then the second slave gets disconnected and the whole system stops and nothing happens anymore. 

Normally, or at least what I am trying to achieve here, is that the master sends every received packet to the second slave using the bt_l2cap_chan_send() function. 

This is my receive callback function where I am sending the data to the second slave. When I use a delay of 1 ms, the master is able to send every packet to the second slave, which is the behaviour I am wishing for. 
[image]

carlescufi, 2020-08-25:
@sarazalabany could you manage to get a backtrace of the device that gets the bus fault? At the very least can you run `addr2line` on the 0x00081ce address that triggered the fault?

dleach02, 2020-09-16:
Is this issue only with 2.2?

carlescufi, 2020-11-26:
Closing due to inactivity. Plesae reopen if this is still an issue.

closed 2020-11-26 by carlescufi