zephyrproject-rtos/zephyr #27673: DFU: support mcuboot direct execute-in-place (XIP)

nvlsianpu, 2020-08-19:
**dual slot: execution in place**
[mcuboot direct execute-in-place (XIP) mode](https://github.com/JuulLabs-OSS/mcuboot/pull/739) was recently
introduced. This mode allow to run either image form slot_0 and slot_1. Mcuboot doesn't perform any swap operation - just discovery which image may and should to be run and run it.

- [ ] Update DFU submodules for make them able to operate either on slot_0 and slot_1
- [ ] allow to build application either for slot_0 or slot_1

nvlsianpu, 2020-08-19:
cc @utzig

nvlsianpu, 2020-11-12:
@de-nordic Is working on solution for that.

danie007, 2022-02-07:
Hi @nvlsianpu @de-nordic 

We're trying to perform **dual slot: execution in place** software upgrade on NXP i.MX RT 11xx MCU

Is this feature available yet? could you point us to the documentation?

---

CC @tkmozhi

de-nordic, 2022-02-07:
> 
> We're trying to perform **dual slot: execution in place** software upgrade on NXP i.MX RT 11xx MCU
> 
> Is this feature available yet? could you point us to the documentation?
> 

Half available: The Direct-XIP is supported by mcuboot, but DFU from Zephyr is not out-of-the-box solution as application needs to be built for specific slot it will be running from and there is no direct support for building pair of applications as candidates for both slots; the mcumgr supports rejecting application that was not built for DFU available slot with special TLV added to the header of binary and Kconfig options enabled at build.

nvlsianpu, 2022-02-07:
@danie007 
@de-nordic Describes situation very well. Right now you might use the direct-xip mode and run the zephyr application, but this requires modification of DTS fixed-partition for building the application for the alternative slot (slot-0 must swap the place with the slot-1). And as Dominik mentioned application can't reject the image for bad slot while it downloading the image.

ExtremeGTX, 2022-04-17:
I think this has a dependency on the Hardware to support address remapping, some controllers are able to remap some physical XIP address range to a CPU address range (this is not related to MMU).

I have used mcuboot before in a similar situation where the decision of mcuboot about which image to be booted is passed to the do_boot function which will check if a remapping is needed then jump to the image.

I think ST, NXP, ESP32 has such remapping feature to support dual image, but AFAIK mcuboot don't make use of such HW feature.