zephyrproject-rtos/zephyr #27675: i2c: complete and stabilize support for "slave" i2c devices

pabigot, 2020-08-19:
The current i2c_slave driver API is marked experimental and is generally undocumented.  It is also unsuitable for DMA-based solutions like [Nordic SPIS](https://github.com/zephyrproject-rtos/zephyr/issues/21445#issuecomment-566612629).

Review the existing API and the requirements of other hardware and provide an API that can be fully documented, tested, and implemented on any platform that supports this functionality.

teburd, 2024-02-12:
Renamed slave to target, conforming with the updated terminology. The target APIs are a bit limiting today as they..

1. expect single byte transfers and callbacks between
2. require callbacks (which are not useful from usermode threads at all).

That said the API is documented, there are users in the tree, and its been around awhile. So I think while the current API may not necessarily be "complete" or ideal, it is well past the experimental stage.

dpkristensen, 2024-02-26:
@teburd I am attaching my implementation of the TWIS driver for reference if you want, as well as a driver I wrote to use it.  It is kind of a hack as it only supports "buffer mode" instead of the normal byte-wise API, but it does provide a path for using TWIS on the Nordic platform.

[i2c-nordic-twis.zip](https://github.com/zephyrproject-rtos/zephyr/files/14412377/i2c-nordic-twis.zip)

To instantiate, it requires a custom driver provided by the build; the DTS changes for nRF5340 processors look like this:
```
&i2c1 {
	// Two-Wire Interface Slave with EasyDMA
	// SDA: P1.02
	// SCL: P1.03
	compatible = "nordic,nrf-twis";

	foo-driver@4c {
		compatible = "my-company,i2c-target-command";
		status = "okay";

		reg = <0x4c>;
	};
};
```

To initialize the handler layer, it is necessary to hook up the driver:
```
static i2c_target_command_context_t const cmd_ctx = {
    .dev = DEVICE_DT_GET(DT_NODELABEL(weapon_target)),
    .command_handler = _command_handler,
    .stop_handler = _stop_handler,
};

static uint8_t rsp_buffer[30]; // Max response size
static i2c_target_command_response_context_t rsp_ctx = {
    .rsp_buf = rsp_buffer;
};

i2c_target_command_init(&cmd_ctx, &rsp_ctx);
```

The command handler should set the data and size in rsp_ctx for the next read transaction from the host.

dpkristensen, 2024-02-26:
Since the protocol and function of the target is device-specific, there's not really any generic functionality to add; it is up to the implementer to make the device do what it needs to.

You perform an I2C write transaction to send a command with no data, and an I2C write+read transaction to write a command that expects some data in return (it has to be available immediately because it is in IRQ context, so it may require a signaling mechanism to say when data is ready).