zephyrproject-rtos/zephyr #27692: Allow to select between advertising packet/scan response for BT LE device name

kurddt, 2020-08-20:
Currently when starting legacy advertising and setting `BT_LE_ADV_OPT_USE_NAME` option, the device name is set inside the scan response if one is provided when calling `bt_le_adv_start` else inside the advertising packet.

So there is not user-friendly way to send the same in the advertising packet if `BT_LE_ADV_OPT_USE_NAME` and scan response are used.

It would be nice to have an additional flag to select where to put the name

A workaround to achieve the desired result is to no use `BT_LE_ADV_OPT_USE_NAME` and set explicitly the device name in the advertising packet

joerchan, 2020-12-08:
@jhedberg Currently bt_set_name would clear the rest of the Scan Response data.
Should we perhaps change that and instead document that the application will have to toggle advertising (in order to provide the scan response data pointer again) when setting the name this way and using the USE_NAME option?

Should we add the option to select the advertising name in the advertising packet instead? I remember you had some opinions on this.
If we don't want to do this we should close this issue.

jhedberg, 2020-12-08:
@joerchan I don't have a strong opinion either way wrt `bt_set_name()`, however that seems orthogonal to this particular issue. My gut feeling is that putting the name in the advertising data rather than the scan response is such a minority use case that we can require apps to just do the manual work in that case. I'm open to be convinced to change my mind, however 😄

PR #32543 (2021-02-23): Bluetooth: host: Don't auto-update device name and document behavior
Don't auto-update the device name in the advertising data as that
would clear any other scan-response data set by the application.
Document the behavior of the advertise with device name option
and the need to update data by the application to make sure
advertising data does not get cleared.

Fixes: #27692
closed 2021-02-23 by carlescufi