zephyrproject-rtos/zephyr #27693: Crash on ARM when BT LE scan response packet too big

kurddt, 2020-08-20:
**Describe the bug**
When the legacy BT LE scan response is too big an bus fault is thrown on arm 

**To Reproduce**
Modify the BLE_beacon example in order to:
* Change toe advertising option to `BT_LE_ADV_CONN_NAME`
* Use synchronous BT initialization:
 change `bt_enable(bt_ready)` to 
```
bt_enable(NULL);
bt_ready
```
* Add more data in the scan response `sd[]`
* Use dynamic name `CONFIG_BT_DEVICE_NAME_DYNAMIC=y`

**Expected behavior**
`bt_le_adv_start` should return an error

**Impact**
This issue is annoying but produced by a wrong user input

**Logs and console output**
```
[00:00:02.828,857] <err> os: r0/a1:  0x00000003  r1/a2:  0x00000001  r2/a3:  0x00000001
[00:00:02.828,857] <err> os: r3/a4:  0x0000139d r12/ip:  0x00000000 r14/lr:  0x0000629d
[00:00:02.828,857] <err> os:  xpsr:  0x61000011
[00:00:02.828,857] <err> os: Faulting instruction address (r15/pc): 0x000062a8
[00:00:02.828,887] <err> os: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
[00:00:02.828,887] <err> os: Fault during interrupt handling

[00:00:02.828,887] <err> os: Current thread: 0x20001258 (unknown)
[00:00:09.294,616] <err> os: Halting system
```

When using my application instead of the BLE_beacon, the error was a bus fault instead of a faulting instruction address

joerchan, 2020-09-16:
@kurddt Can you please provide some more details, I'm not able to reproduce this problem.
Which board?
Which zephyr commit or tag?
What exactly did you add to sd (or how many bytes)?
What is CONFIG_BT_DEVICE_NAME set to (or how many characters)?

Could be that you hit the exact place where there is an off-by-one error, but atleast for me it either starts, or it fails to start, with a log "Too big advertising data"

kurddt, 2020-09-17:
The board is a custom one, with a NRF52832 chip on it. 
I'm using zephyr-v2.3.0 version
CONFIG_BT_DEVICE_NAME_DYNAMIC=y,

joerchan, 2020-09-17:
Update from slack discussion. Bug appears on 2.3.0 and master when the scan response data is of a length 30 or 31.
This is due to the device name shortening handling, which in this case will have an integer underflow of the shortened length field.

PR #28468 (2020-09-19): Bluetooth: host: Fix bug in device name shortening handling
Fix bug in device name shortening handling leading to memory
corruption.
This is triggered by an underflow in the length field of the shortened
name when set_data_len + 2 > set_data_len_max.

Fixes: #27693
commit 0fe62c63 (2020-09-19): Bluetooth: host: Fix bug in device name shortening handling

Fix bug in device name shortening handling leading to memory
corruption.
This is triggered by an underflow in the length field of the shortened
name when set_data_len + 2 > set_data_len_max.

Fixes: #27693
closed 2020-09-19 by carlescufi