zephyrproject-rtos/zephyr #27697: Add support for passing -nogui 1 to J-Link Commander on MS Windows

henrikbrixandersen, 2020-08-20:
**Is your enhancement proposal related to a problem? Please describe.**
PR #27617 introduced support for detecting the version of the installed Segger J-Link Commander in order to pass `-nogui 1` command line parameter to supported versions (> 6.80) for suppressing GUI windows showing flash progress.

The check introduced, however, does not work with Segger J-Link Commander on Microsoft Windows.

**Describe the solution you'd like**
Find out why the introduced check does not work on Windows and fix it.

**Describe alternatives you've considered**
Accept the Segger J-Link Commander on Windows present... windows ... when flashing.

JakeMercer, 2021-04-23:
I think I may have found this issue on Linux:
```
jmercer@mercury:~/Development/DUTCHess$ JLinkExe fds 
SEGGER J-Link Commander V7. 0a (Compiled Apr 16 2021 15:36:25)
DLL version V7.00a, compiled Apr 16 2021 15:36:09

Unknown command line option fds.
```

Version string for this version of JLink Commander has a space instead of the digit expected by the regex which parses it (from _jlink.py_):
`\s+V([.0-9]+)[a-zA-Z]*\s+`

This results in "7." being matched which does not conform to the PEP440 format expected by the packaging python library; so this code results in the "7." falling back to a type of `packaging.version.LegacyVersion` instead of `packaging.version.Version`
```
def supports_nogui(self):
    ver = self.read_version()
    # -nogui was introduced in J-Link Commander v6.80
    return version.parse(ver) >= version.parse("6.80")
```
Which, when compared, are subject to the following stanza in the packaging documentation:

> This class implements the previous de facto sorting algorithm used by setuptools, however it will always sort as less than a Version instance.

henrikbrixandersen, 2021-05-03:
Can anybody (@mhelm, @mbolivar-nordic?) confirm If this was fixed with #34660?

closed 2021-05-06 by henrikbrixandersen

mbolivar-nordic, 2021-05-06:
I just retested. It does pass -nogui 1 on flash and that does work.

I also passes -nogui on debug, but I still get a GUI. There's nothing I can see wrong with what west is doing, though.

henrikbrixandersen, 2021-05-06:
Oh. I didn't test debug. I'll leave this as closed, though.

Can't see anything wrong with the west runner either. Thanks.