zephyrproject-rtos/zephyr #27734: vl53l0x driver gives wrong offset calibration value

jrananth1032, 2020-08-24:
**Describe the bug**
We are vl53l0x proximity sensor in our product. We have cover glass to protect it from dust and other things.So after installing cover glass we need to do calibrations as per datasheet. We got proper values for SPAD and reference(temperature) calibration but offset and xtalk calibration values or looks like wrong.I am thinking that this line may have problem https://github.com/zephyrproject-rtos/hal_st/blob/5b3ec3e182d4310e8943cc34c6c70ae57d9711da/sensor/vl53l0x/api/core/src/vl53l0x_api_calibration.c#L258, Here StoredMeanRangeAsInt will become 0.Can anyone help me to solve this,Thanks in advance.

erwango, 2020-08-24:
@jrananth1032 the line you mention is directly taken from ST Driver library and is not supported inside Zephyr project.
I'd propose to raise your question on https://community.st.com/

jrananth1032, 2020-08-24:
okay thanks

erwango, 2020-08-24:
@jrananth1032, since the issue does not depends directly on Zephyr code base, I'm closing the point.
Please feel free to reopen if needed

closed 2020-08-24 by erwango