zephyrproject-rtos/zephyr #27735: Enable DT-based sanity-check test filtering

ioannisg, 2020-08-24:
**Is your feature request related to a problem? Please describe.**
In sanity check we have efficient way to whitelist or to exclude (filter out) tests/test-suites on certain platforms based on
- Kconfig switches
- ARCHitecture or platform target names
- required RAM / Code footprint (with respect to platform capabilities)

We would like to have additional filtering options based on HW capabilities of the platforms that are expressed via Device Tree nodes. This will prevent tests from running and, inevitably, failing, on platform which do not have the required HW capabilities to run those tests.

**Describe the solution you'd like**

Just to give an example: in some configurations, certain User-space test-suites require a minimum number of ARM MPU regions to execute properly. The number of MPU regions is described by a DTS node in the SoC .dtsi header. We would like sanity check to detect and exclude such tests from running on platforms that do not have the minimum required number of MPU regions.

ioannisg, 2020-08-24:
@nashif @PerMac FYI
If there is already a related discussion about this, please, comment and close the issue.

gmarull, 2022-08-11:
We have now quite a few of `dt_` helpers that can be used in `filter`. Closing, re-open if needed.

closed 2022-08-11 by gmarull