zephyrproject-rtos/zephyr #27745: Zephyr with host stack and hci driver only ?

icephyr, 2020-08-24:
Hello guys, I met a problem during development and hope anyone can help.

I'd like to use an i.mxRT1050 running zephyr as a host, the RT1050 connects with an BLE chip nRF52832 by uart. RT1050 will collect data from sensors, connect and send data to the Gateway by nRF52832, so the RT1050 shall support BLE Host stack and uart driver. The RT1050 will control the nRF52832 by HCI command.  I wonder if such mode is supported by zephyr now，I didn't find any samples in bluetooth folder like this.

carlescufi, 2020-08-24:
@icephyr just build `samples/bluetooth/hci_uart` and it will give you what you need.
This is described here:
https://docs.zephyrproject.org/latest/samples/bluetooth/hci_uart/README.html

closed 2020-08-24 by carlescufi

icephyr, 2020-08-25:
> @icephyr just build `samples/bluetooth/hci_uart` and it will give you what you need.
> This is described here:
> https://docs.zephyrproject.org/latest/samples/bluetooth/hci_uart/README.html

Hi carlescufi, thanks a lot for your quick reply. One question about hci_uart sample is it is suitable for the nRF52832 as a controller, as for the RT1050 playing the host role, also run hci_uart sample too ?