zephyrproject-rtos/zephyr #27753: drivers: sensor: lis2dh: compilation issue struct lis2dh_config' has no member named 'spi_conf'

ABOSTM, 2020-08-24:
**Describe the bug**
While pushing PR #27013,  Buildkite highlight a compilation issue in lis2dh.

Note: Even if issue is not directly linked to the PR #27013,
the PR highlight this issue because it enabled adc for stm32f3_disco, which is a condition (depends on) for sanitycheck 
 to compile for this board.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p auto -b stm32f3_disco tests/drivers/build_all/ -- -DCONF_FILE=sensors_trigger_i_z.conf
2. See error

**Expected behavior**
Compilation successful.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
```
FAILED: zephyr/drivers/sensor/lis2dh/CMakeFiles/drivers__sensor__lis2dh.dir/lis2dh_spi.c.obj
C:\gnuarmemb\bin\arm-none-eabi-gcc.exe -DBUILD_VERSION=zephyr-v2.3.0-2029-g5449436515cd -DCORE_CM4 -DHSE_VALUE=8000000 -DKERNEL -DSTM32F303xC -DUSE_FULL_LL_DRIVER -DUSE_HAL_DRIVER -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR_SUPERVISOR__ -D__ZEPHYR__=1 -I../include -Izephyr/include/generated -I../soc/arm/st_stm32/stm32f3 -I../drivers -I../soc/arm/st_stm32/common -I../subsys/testsuite/include -IC:/zephyr/zephyr_project/modules/hal/cmsis/CMSIS/Core/Include -IC:/zephyr/zephyr_project/modules/hal/st/sensor/stmemsc/iis3dhhc_STdC/driver -IC:/zephyr/zephyr_project/modules/hal/st/sensor/stmemsc/lis2mdl_STdC/driver -IC:/zephyr/zephyr_project/modules/hal/st/sensor/stmemsc/lsm6dso_STdC/driver -IC:/zephyr/zephyr_project/modules/hal/stm32/stm32cube/stm32f3xx/soc -IC:/zephyr/zephyr_project/modules/hal/stm32/stm32cube/stm32f3xx/drivers/include -IC:/zephyr/zephyr_project/modules/hal/stm32/stm32cube/stm32f3xx/drivers/include/Legacy -isystem ../lib/libc/minimal/include -isystem c:/gnuarmemb/bin/../lib/gcc/arm-none-eabi/9.2.1/include -isystem c:/gnuarmemb/bin/../lib/gcc/arm-none-eabi/9.2.1/include-fixed -Os -imacrosC:/zephyr/zephyr_project/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -imacrosC:/zephyr/zephyr_project/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-address-of-packed-member -Wno-pointer-sign -Wpointer-arith -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=C:/zephyr/zephyr_project/zephyr/tests/drivers/build_all=CMAKE_SOURCE_DIR -fmacro-prefix-map=C:/zephyr/zephyr_project/zephyr=ZEPHYR_BASE -fmacro-prefix-map=C:/zephyr/zephyr_project=WEST_TOPDIR -ffunction-sections -fdata-sections -std=c99 -nostdinc -MD -MT zephyr/drivers/sensor/lis2dh/CMakeFiles/drivers__sensor__lis2dh.dir/lis2dh_spi.c.obj -MF zephyr\drivers\sensor\lis2dh\CMakeFiles\drivers__sensor__lis2dh.dir\lis2dh_spi.c.obj.d -o zephyr/drivers/sensor/lis2dh/CMakeFiles/drivers__sensor__lis2dh.dir/lis2dh_spi.c.obj   -c C:/zephyr/zephyr_project/zephyr/drivers/sensor/lis2dh/lis2dh_spi.c
C:/zephyr/zephyr_project/zephyr/drivers/sensor/lis2dh/lis2dh_spi.c: In function 'lis2dh_raw_read':
C:/zephyr/zephyr_project/zephyr/drivers/sensor/lis2dh/lis2dh_spi.c:26:41: error: 'const struct lis2dh_config' has no member named 'spi_conf'
   26 |  const struct spi_config *spi_cfg = &cfg->spi_conf;
      |                                         ^~
C:/zephyr/zephyr_project/zephyr/drivers/sensor/lis2dh/lis2dh_spi.c: In function 'lis2dh_raw_write':
C:/zephyr/zephyr_project/zephyr/drivers/sensor/lis2dh/lis2dh_spi.c:72:41: error: 'const struct lis2dh_config' has no member named 'spi_conf'
   72 |  const struct spi_config *spi_cfg = &cfg->spi_conf;
```

**Environment (please complete the following information):**
 - OS:  Windows
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA: 5449436515cd2098abeeda3bb22ff6b88ae0c384

**1st Analysist**
In file lis2dh.h, the member `spi_conf ` of struct `lis2dh_config` is defined under compilation switch.
And it is defined only if sensor is okay in spi DT node, but only if sensor is not okay in i2c DT node.
```
struct lis2dh_config {
	char *bus_name;
	int (*bus_init)(struct device *dev);
#if DT_ANY_INST_ON_BUS_STATUS_OKAY(i2c)
	uint16_t i2c_slv_addr;
#elif DT_ANY_INST_ON_BUS_STATUS_OKAY(spi)
	struct spi_config spi_conf;
#if DT_INST_SPI_DEV_HAS_CS_GPIOS(0)
	const char *gpio_cs_port;
	uint8_t cs_gpio;
	uint8_t cs_gpio_flags;
#endif /* DT_INST_SPI_DEV_HAS_CS_GPIOS(0) */
#endif /* DT_ANY_INST_ON_BUS_STATUS_OKAY(spi) */

};
```

Is it allowed to have 2 sensors, one on i2c the other on spi ? 
Whatever, in stm32f3_disco board, there is a sensor defined in i2c DT node.
Thus it fails when compilling spi source:  lis2dh/lis2dh_spi.c

Maybe in such case both `i2c_slv_addr` and `spi_conf` should be defined at the same time ?
like that:
```
struct lis2dh_config {
	char *bus_name;
	int (*bus_init)(struct device *dev);
#if DT_ANY_INST_ON_BUS_STATUS_OKAY(i2c)
	uint16_t i2c_slv_addr;
#endif /* DT_ANY_INST_ON_BUS_STATUS_OKAY(i2c) */
#if DT_ANY_INST_ON_BUS_STATUS_OKAY(spi)
	struct spi_config spi_conf;
```
This at least solve the compilation issue.

ABOSTM, 2020-08-24:
^^ @avisconti

ABOSTM, 2020-08-25:
```
Maybe in such case both i2c_slv_addr and spi_conf should be defined at the same time ?
like that:

struct lis2dh_config {
	char *bus_name;
	int (*bus_init)(struct device *dev);
#if DT_ANY_INST_ON_BUS_STATUS_OKAY(i2c)
	uint16_t i2c_slv_addr;
#endif /* DT_ANY_INST_ON_BUS_STATUS_OKAY(i2c) */
#if DT_ANY_INST_ON_BUS_STATUS_OKAY(spi)
	struct spi_config spi_conf;
This at least solve the compilation issue.
```
In fact, it is not a good solution: I face another compilation error. It is looking for spi-max-frequency property, but within i2c DT node.

avisconti, 2020-08-31:
Just back from vacations and having a big agenda of stuffs to accomplish. Will take a look asap.

avisconti, 2020-09-07:
Still not working on it, but I quickly took a look at it and seems to me that I may see what has been done for bme280 (8aadf895) for multi-instance implementation.

carlescufi, 2020-09-08:
@avisconti thanks, if you need additional info @mbolivar-nordic is in the #devicetree channel in Slack.

PR #28227 (2020-09-14): drivers: sensor: lis2dh: make the driver multi-instance
Make this driver multi-instance and use the new API.
Fixes #27753.

Tested on x_nucleo_iks01a3 shield for both I2C and SPI. I used two different shields where the DIL24 was configured for I2C (first shield) and SPI (second shield).
commit 4ea9a8bd (2020-09-14): drivers: sensor: lis2dh: make the driver multi-instance

Make this driver multi-instance and use the new API.
Fixes #27753.
closed 2020-09-14 by MaureenHelm