zephyrproject-rtos/zephyr #27767: Merge the MQTT-SN library into the OpenThread submodule

oubotong, 2020-08-25:
Currently, I can't find a way to use MQTT-SN protocol within Zephyr. There is no library to use.
However, I found a repo which embed MQTT-SN into OpenThread: https://github.com/kyberpunk/openthread-mqttsn which is based on **paho** library: https://github.com/eclipse/paho.mqtt-sn.embedded-c

And I tried to port this MQTT-SN into the specific OpenThread which Zephyr is using as a module and tested it which works well.(with `zephyr/CmakeList.txt` set up).'

Here is the repo which I port MQTT-SN with the Openthread used by Zephyr: https://github.com/oubotong/openthread
You just need to use this replace the one created by `west init`. They have the same commit number.

To test it, you need to setup a Border_Router with OpenThread and set it up as a broker. Then you can refer to this link for testing MQTT-SN: https://github.com/kyberpunk/openthread/tree/979b4eecd8a414e734fb568cb53d2ba42d1cfc07

Now the problem is, within the OpenThread by running command like `make -f example/Makefile-nrf52840 MQTT=1 JOINER=1`, the output binary works well. However if I use this in my Zephyr application, my end device can only sent out `SEARCHGW` message. The `CONNECT` can't received by the broker. So I hope some coding experts could take a try at this porting and figure out how to use it in an Zephyr application. 

I found that there is both OpenThread in `zephyr-project/modules/lib` and in `zephyr-project/zephyr/subsys/net/l2`. I am wondering if there is a way to only use the outside OpenThread module instead of the inner one.

rlubos, 2020-09-01:
Hi, just to clarify `zephyr-project/modules/lib/openthread` is a copy of the OpenThread repository, following the upstream repo (https://github.com/openthread/openthread) with a minor Cmake extension, allowing to integrate OpenThread build system with Zephyr.  `zephyr-project/zephyr/subsys/net/l2/openthread` is the Zephyr integration of OpenThread - this allows to use OpenThread as a Zephyr L2, and provide OpenThread platform based on Zephyr APIs ( `zephyr-project/zephyr/subsys/net/lib/openthread/platform`.

Regarding `MQTT-SN` support in OpenThread, given that the upstream OT implementation doesn't have this feature (it's developed on a bit outdated fork) it's rather unlikely we'll introduce it into Zephyrs OpenThread copy. I don't want to deviate from the upstream repository, as this will cause problems with OT updates in the future. It's the OpenThread community that would need to be convinced to accept this feature in the first place.

rlubos, 2024-02-09:
Closing, as:
1. There are no plans to integrate 3rd party OpenThread forks into Zephyr
2. Zephyr now ships its own MQTT-SN library: https://docs.zephyrproject.org/latest/connectivity/networking/api/mqtt_sn.html

closed 2024-02-09 by rlubos