zephyrproject-rtos/zephyr #27771: iotdk: cpu_stats function doesn't work as expected

maksimmasalski, 2020-08-25:
**Describe the bug**
cpu_stats profiling function doesn’t work as expected, it always report cpu usage 100 even if cpu in idle thread for a long time. It looks like the cpu_stats hooks does not have a chance to count the switch in of idle thread (stats_hw_tick.idle always 0). 

```
Hello World! iotdk
CPU usage: 100
CPU usage: 100
CPU usage: 100
CPU usage: 100
CPU usage: 100
```
It always report cpu usage 100%, and global variable stats_hw_tick.idle is 0, but I confirmed cpu can enter idle

**To Reproduce**
1. Use latest master c64e0ce923cde54415d203e447a1dc62894ed2b0
2. SDK 0.11.4
3. Go to samples/hello_world
4. Add to prj.conf configuration for turning on cpu_stats:
```
CONFIG_TRACING=y
CONFIG_TRACING_CPU_STATS=y
CONFIG_TRACING_CPU_STATS_LOG=y
```
5. west build -p -b iotdk samples/hello_world/
6. west flash
7. View the output

**Expected behavior**
When system idle, the cpu usage reported should not be always 100%
My expectation
```
Hello World! iotdk
CPU usage: 0
CPU usage: 0
CPU usage: 0
CPU usage: 0
CPU usage: 0
```

**Impact**
Wrong cpu_stats function execution on the arc board.

**Environment (please complete the following information):**
 - OS: Linux Ubuntu 18.04.4 LTS
 - Toolchain Zephyr SDK 0.11.4
 - Commit SHA c64e0ce923cde54415d203e447a1dc62894ed2b0

nashif, 2020-08-31:
Here we are not recording switched_in/switched_out correctly for all threads and cpu stats depends on such information. switched_in might need to move to the arch context switch code and not be in the c code.

jocelyn-li, 2020-09-02:
@evgeniy-paltsev I am interested with this fix. Do you have timeline for the fix?

PR #28001 (2020-09-03): tracing: swap: bug fix and enhancement for ARC
Describe in #27771 cpu_stats profiling function doesn’t work as expected. we need to do some fixes.

For regular context switch: 
move switched_in into the arch context switch assembly code, 
which will correctly record the switched_in information.

For irq context switch: 
add switched_in/switched_out for context switch in irq exit.

Fixes: #27771
PR #28426 (2020-09-23): tracing: swap: bug fix and enhancement for ARC
Describe in #27771 cpu_stats profiling function doesn’t work as expected. we need to do some fixes.

For regular context switch:
move switched_in into the arch context switch assembly code,
which will correctly record the switched_in information.

For irq context switch:
add switched_in/switched_out for context switch in irq exit.

Fixes: #27771
closed 2020-09-03 by carlescufi