zephyrproject-rtos/zephyr #27772: Use LVGL's Kconfig

kisvegabor, 2020-08-25:
Hi,

At LVGL we are planning to add a Kconfig file to the main repo but we are not sure if it will harm the existing Kconfig based projects. 

Could you comment on what your requirements are and what we should consider in order to allow Zephyr to use LVGL's Kcofig file or - if it's not possible - to make sure it won't bother your Kconfig file?

cc @c47d @BFFonseca

gmarull, 2020-08-26:
As far as I know Zephyr doesn't use external modules Kconfig files directly, which makes sense to me. I think that if LVGL moves to Kconfig Zephyr could definitely benefit from it, Kconfig files could probably be re-used with some patches for Zephyr specific options (e.g. display device name). I think that the main Zephyr requiremements now would be that default enabled features are minimal and Kconfig options prefixed (Zephyr prefixes all LVGL options with `LVGL_`, `LV_` could probably be acceptable). If same Kconfig names could be shared `lv_conf.h` maintenance cost would be reduced, assuming Kconfig settings are automatically used.

kisvegabor, 2020-08-26:
Thanks, @gmarull! Glad to hear that you like the idea.

I wonder if you can simply add LVGL's Kconfig with `rsource` similarly you do it [here](https://github.com/zephyrproject-rtos/zephyr/blob/202ca01dcbe760b825f550df4d8411ad09f34df0/lib/gui/lvgl/Kconfig#L545).

The minimal configuration by default can be an issue because for LVGL users it's more straightforward to have all features enabled which are required to run the examples (it means all widgets). Probably unused features will be dropped during linking anyway. 

I'm not an expert of Kconfig but maybe you can suggest a way to alter the default values. E.g.
```
  default y if !LVGL_CONFIG_MINIMAL
```

gmarull, 2020-08-26:
> Thanks, @gmarull! Glad to hear that you like the idea.
> 
> I wonder if you can simply add LVGL's Kconfig with `rsource` similarly you do it [here](https://github.com/zephyrproject-rtos/zephyr/blob/202ca01dcbe760b825f550df4d8411ad09f34df0/lib/gui/lvgl/Kconfig#L545).

We'd need to have a copy in Zephyr repository, Kconfig from external modules can't be directly included. This is because Zephyr Kconfig needs to be operational even if some modules are not available. However, this doesn't prevent re-use, files would be synchronized (and possibly patched with necessary tweaks) when Zephyr upgrades to new versions of LVGL. License compatibility also needs to be taken into account if files are copied.

> The minimal configuration by default can be an issue because for LVGL users it's more straightforward to have all features enabled which are required to run the examples (it means all widgets). [...]

I'd be fine using LVGL default set, in the end non-used stuff should be pruned from the resulting binary and simplifies running LVGL official demos. `LVGL_CONFIG_MINIMAL` sounds like a good idea.

kisvegabor, 2020-08-26:
> We'd need to have a copy in Zephyr repository, Kconfig from external modules can't be directly included. This is because Zephyr Kconfig needs to be operational even if some modules are not [...]

OK, great!

> I'd be fine using LVGL default set, in the end non-used stuff should be pruned from the resulting binary and simplifies running LVGL official demos.

It's also great. As you mentioned in the worst case you can patch some default values.

vanwinkeljan, 2020-08-27:
Sound like a good idea and as @gmarull said if we line up the config names there is no need in maintaining a version of `lv_conf.h` in zephyr.

> .. in the end non-used stuff should be pruned from the resulting binary ..

That is definitely not true some feature will be included in functions that can not be removed and will result in a bigger code base. But if a config such as `LVGL_CONFIG_MINIMAL` would be available we can just set it to default.

C47D, 2020-08-27:
Hi all,

The current Kconfig can be found [here](https://github.com/lvgl/lv_port_esp32/blob/master/components/lvgl/Kconfig). And the lv_conf.h file can be found [here](https://github.com/lvgl/lv_port_esp32/blob/master/components/lvgl/lv_conf.h). We can agree on the config names before the Kconfig file is summited to the main LVGL repo.

kisvegabor, 2020-08-28:
@vanwinkeljan 
> That is definitely not true some feature will be included in functions that can not be removed and will result in a bigger code base.

True. I'll take care to use something like `LVGL_CONFIG_MINIMAL`.

kisvegabor, 2020-08-28:
@C47D 
> We can agree on the config names before the Kconfig file is summited to the main LVGL repo.

I vote for using the exact same names as in `lv_conf.h` (e.g. `LV_USE_BTN`) to maintain compatibility with `lv_conf.h`.

C47D, 2020-08-28:
> I vote for using the exact same names as in lv_conf.h (e.g. LV_USE_BTN) to maintain compatibility with lv_conf.h.

Will open an issue on the `lv_port_esp32` repo to update the symbol names, currently they start with `LVGL`.

carlescufi, 2021-12-20:
Closing this are resolved, but feel free to reopen @kisvegabor

closed 2021-12-20 by carlescufi

kisvegabor, 2021-12-20:
I've just asked about this topic in the [new PR](https://github.com/zephyrproject-rtos/zephyr/pull/41336#issuecomment-998005106). Let's discuss it there.

carlescufi, 2021-12-20:
Reopening this @kisvegabor, I think it makes sense to keep it open for now.

kisvegabor, 2021-12-21:
Thanks. Although, unfortunately I can comment much on  "how would it be best to do it in Zephyr". However, if you have some requirements from LVGL's side to make your life easier, pleas let me know.