zephyrproject-rtos/zephyr #27785: memory domain arch implementation not correct with respect to SMP on ARC

andrewboie, 2020-08-25:
**Describe the bug**
Consider an SMP system with two CPUs.

- CPU 0 is running a supervisor thread A which makes some API calls to configure a memory domain X
- CPU 1 is running a user thread B which is a member of memory domain X

With the current code, the memory domain API changes made by thread A will not cause the MPU on CPU 1 to be reconfigured, and thread B retains the old memory domain configuration until it is context switched out and then in again.

**To Reproduce**
Found via code inspection.

**Expected behavior**
arch_mem_domain_* API invocations on some memory domain X synchronously reprogram the MPU on other CPUs if any of the other CPUs are running a user thread in memory domain X.

If another CPU is running a supervisor thread that belongs to memory domain X, this can be ignored, as memory domains only apply to user threads.

**Impact**
I think this is low. Memory domains are typically set up at application initialization time, reprogramming memory domain properties as part of an application's normal state machine is an odd use-case.

andrewboie, 2020-08-26:
note: I'm going to write a test case for this as I'm not sure everything is in order on x86 either.

ruuddw, 2020-08-26:
Noted. We don't enable/support Userspace on ARC HS yet, there might be other issues as well with SMP and userspace.

andrewboie, 2020-08-26:
> Noted. We don't enable/support Userspace on ARC HS yet

ohh! misunderstanding on my part.
Feel free to close unless you want to leave this open as a reminder.

closed 2020-08-26 by andrewboie

abrodkin, 2020-10-13:
@andrewboie any hints on how that should or could be done properly? Indeed if another core is running cooperative thread (i.e. we cannot interrupt it) how can we re-program its MPU or MMU if it's an essential part of the core, i.e. only core itself may access its MPU or MMU, but not its peers in the cluster.

andrewboie, 2020-10-13:
> @andrewboie any hints on how that should or could be done properly? Indeed if another core is running cooperative thread (i.e. we cannot interrupt it) how can we re-program its MPU or MMU if it's an [...]

I expect you'll want something like an inter-processor interrupt to cause a page table or MPU register re-load.

abrodkin, 2020-10-14:
> > @andrewboie any hints on how that should or could be done properly? Indeed if another core is running cooperative thread (i.e. we cannot interrupt it) how can we re-program its MPU or MMU if it's an essential part of the core, i.e. only core itself may access its MPU or MMU, but not its peers in the cluster.
> 
> I expect you'll want something like an inter-processor interrupt to cause a page table or MPU register re-load. Co-op threads don't disable interrupts.

Right, this is possible, yet we'll need to interrupt the [cooperative] task currently being executed to re-program PMU under its feet, right? And that was exactly my concern - may we interrupt cooperative task in that case?

ruuddw, 2020-10-14:
An IPI interrupt would work, cooperative tasks can be interrupted (when they haven't locked interrupts).
If the interrupt locking is an issue, another solution would be to ensure/enforce no other task is running when changing memory domain settings. Or, at least no other task that is touched by the change (in the same memory domain).

andrewboie, 2020-10-14:
> Right, this is possible, yet we'll need to interrupt the [cooperative] task currently being executed to re-program PMU under its feet, right? And that was exactly my concern - may we interrupt cooperative task in that case?

To me this is as simple as sending an IPI and letting the IPI handler (running like any other ordinary interrupt) reprogram the MPU whenever it gets it.

I believe there's no requirement that we need to worry about interrupt locking, if IRQs are disabled on the other CPU for any reason just handle it once unlocked.

andrewboie, 2020-10-14:
(I have do something similar for TLB shootdowns on MMU systems like x86_64 btw, was working on the code today)