zephyrproject-rtos/zephyr #27791: DT_DRV_COMPAT in spi_flash_w25qxxdv.c named incorrectly

simon-iversen, 2020-08-26:
**Describe the bug**
There is a bug in the driver [zephyr\drivers\flash\spi_flash_w25qxxdv.c](https://github.com/zephyrproject-rtos/zephyr/blob/94721b316f634c3a4c24f7a839ab87363248d0d2/drivers/flash/spi_flash_w25qxxdv.c). [DT_DRV_COMPAT](https://github.com/zephyrproject-rtos/zephyr/blob/94721b316f634c3a4c24f7a839ab87363248d0d2/drivers/flash/spi_flash_w25qxxdv.c#L7) is set to winbond_w25q16, which will make it look for the wrong DTS definitions. It should instead be set to jedec_spi_nor. The additional definition of DT_DRV_COMPAT in [zephyr\drivers\flash\spi_flash_w25qxxdv.h](https://github.com/zephyrproject-rtos/zephyr/blob/94721b316f634c3a4c24f7a839ab87363248d0d2/drivers/flash/spi_flash_w25qxxdv.h#L16) should also be removed

**To Reproduce**
Steps to reproduce the behavior:
1. Download and extract the uploaded file spi_flash
2. Build it with NCS v1.3.0: `west build -d build_nrf52dk_nrf52832 -b nrf52dk_nrf52832`  

**Expected behavior**
Errors due to not finding the spi_flash_w25qxxdv DTS definitions.

**Impact**
Not able to use the driver _zephyr\drivers\flash\spi_flash_w25qxxdv.c_

**Logs and console output**
\`\`\`C:/Users/<..>/ncs/v1.3.0/zephyr/drivers/flash/spi_flash_w25qxxdv.c: In function 'spi_flash_wb_configure':
C:/Users/<..>/ncs/v1.3.0/zephyr/include/devicetree.h:297:40: error: 'DT_N_INST_0_winbond_w25q16_BUS_P_label' undeclared (first use in this function)
 #define DT_INST(inst, compat) UTIL_CAT(DT_N_INST, DT_DASH(inst, compat))
                                        ^~~~~~~~~
C:/Users/<..>/ncs/v1.3.0/zephyr/include/devicetree.h:1786:38: note: in definition of macro 'DT_CAT'
 #define DT_CAT(node_id, prop_suffix) node_id##prop_suffix
                                      ^~~~~~~
C:/Users/<..>/ncs/v1.3.0/zephyr/include/devicetree.h:1302:31: note: in expansion of macro 'DT_PROP'
 #define DT_BUS_LABEL(node_id) DT_PROP(DT_BUS(node_id), label)
                               ^~~~~~~
C:/Users/<..>/ncs/v1.3.0/zephyr/include/devicetree.h:1295:25: note: in expansion of macro 'DT_CAT'
 #define DT_BUS(node_id) DT_CAT(node_id, _BUS)
                         ^~~~~~
C:/Users/<..>/ncs/v1.3.0/zephyr/include/devicetree.h:1302:39: note: in expansion of macro 'DT_BUS'
 #define DT_BUS_LABEL(node_id) DT_PROP(DT_BUS(node_id), label)\`\`\`

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain and version: NCS v1.3.0

pabigot, 2020-08-26:
You should not be using spi_flash_w25qxxdv.c.  All JEDEC SPI flashes should be supported by `CONFIG_SPI_NOR=y` which selects the `spi_nor.c` driver.

We really need to get that W25 one out of tree.

carlescufi, 2020-08-27:
Closing now that #27802 is merged

closed 2020-08-27 by carlescufi