zephyrproject-rtos/zephyr #27805: samples without README

pabigot, 2020-08-26:
Taking #27130 to the next step: The following samples do not have a README:

* [ ] samples/drivers/led_apa102c_bitbang
* [ ] samples/drivers/lora/receive
* [ ] samples/drivers/lora/send
* [ ] samples/drivers/soc_flash_nrf
* [ ] samples/drivers/spi_flash (#27130)
* [ ] samples/drivers/spi_fujitsu_fram
* [ ] samples/drivers/watchdog
* [ ] samples/net/mdns_responder
* [ ] samples/posix/eventfd
* [ ] samples/sensor/bmg160
* [ ] samples/sensor/fxos8700-hid
* [ ] samples/sensor/grove_light
* [ ] samples/sensor/grove_temperature
* [ ] samples/subsys/console/echo
* [ ] samples/subsys/logging/syst
* [ ] samples/subsys/power/device_pm
* [ ] samples/subsys/shell/shell_module

pabigot, 2020-08-26:
```
find samples/ -name sample.yaml \
| sed -e 's@/[^/]*$@@' \
| while read dir ; do \
  if $(cd $dir && ls | grep -i '^readme' > /dev/null) ; then \
    :  ; \
  else  \
    echo '* [ ]' $dir ; \
   fi \
done \
| sort
```

Nukersson, 2020-09-30:
I think `Good First Issue` would be good for the beginners.

kartben, 2023-02-17:
I've updated the checklist in the initial comment with the current status. I've also made sure samples with a README that's not a README.rst are also listed

Spice-Weasel, 2023-10-19:
I will document the lora samples

kartben, 2023-10-19:
> I will document the lora samples

Very nice, thanks!

kartben, 2023-10-20:
@Spice-Weasel let me know if I can help with anything as you go through this. Thanks again!

Spice-Weasel, 2023-10-20:
@kartben will do. I thought this was lorawan and not lora. 😅 I can document it anyway will try to find some time this weekend

kartben, 2023-12-20:
Note to self (or to anyone else reading this and looking for inspiration of what samples might be of higher priority than others): adding the README to ` samples/subsys/shell/shell_module` would be really nice as it's quite a "key" sample for people interesting in developing shell commands, and having some clarity on what the purpose of the sample is would be really useful (some people have started to incorrectly use it as a place to dump overlays and config files that can be used to try all kinds of shell commands on a board, and that's not what this sample is for).

Spice-Weasel, 2023-12-20:
@kartben thanks. I'll have a look at it!

isaacjerish, 2025-04-30:
I'd like to work on the README for samples/sensor/max44009.

PR #67461 (2024-01-19): samples: doc: Added README for POSIX samples
Added README.rst for eventfd and uname POSIX samples.
Helps fix #27805