zephyrproject-rtos/zephyr #27811: intermittent failure of tests/net/socket/select on qemu_x86

andrewboie, 2020-08-26:
**Describe the bug**
Seeing this occasionally fail when running sanitycheck with high CPU load:

```
SeaBIOS (version rel-1.12.1-0-ga5cab58-dirty-20200625_115407-9426dddc0a1f-zephyr
)
Booting from ROM..*** Booting Zephyr OS build zephyr-v2.3.0-2150-gdd655d7c9163  ***
Running test suite socket_select
===================================================================
START - test_fd_set
 PASS - test_fd_set
===================================================================
START - test_select

    Assertion failed at WEST_TOPDIR/zephyr/tests/net/socket/select/src/main.c:115: test_select: (tstamp >= 30U && tstamp <= 30 + FUZZ is false)

 FAIL - test_select
===================================================================
Test suite socket_select failed.
===================================================================
PROJECT EXECUTION FAILED
```

This is very strange as we have icount enabled, so our tests should be deterministic.

wentongwu, 2020-08-26:
The test has the configuration CONFIG_NETWORKING=y which will disable Qemu icount.

andrewboie, 2020-09-03:
> The test has the configuration CONFIG_NETWORKING=y which will disable Qemu icount.

ohhhhhh makes sense now.
Closing. I have a general idea how to mitigate this.

closed 2020-09-03 by andrewboie