zephyrproject-rtos/zephyr #27813: samples without sample.yaml

pabigot, 2020-08-26:
Parallel to #27805 these are samples that lack a `sample.yaml` so may not be participating in CI as expected.

* [ ] samples/application_development/out_of_tree_driver/hello_world_module/zephyr
* [ ] samples/boards/intel_s1000_crb/dmic
* [ ] samples/subsys/mgmt/osdp
* [ ] samples/testing/integration (has a `testcase.yaml` instead)

These applications are subordinate to ones with a `sample.yaml`, not sure what to do about them.

* [ ] samples/subsys/ipc/ipm_mcux/remote
* [ ] samples/subsys/ipc/openamp/remote

jenmwms, 2020-09-01:
From what I understand, the first on this list (`samples/application_development/out_of_tree_driver/hello_world_module/zephyr`) is satisfied by [`samples/application_development/out_of_tree_driver/sample.yaml`](https://github.com/zephyrproject-rtos/zephyr/blob/master/samples/application_development/out_of_tree_driver/sample.yaml)

jenmwms, 2020-09-01:
For this one: `samples/testing/integration`
From what I can tell, this should remain as testcase.yaml based on the accompanying description in [`doc/guides/test/ztest.rst`](https://github.com/zephyrproject-rtos/zephyr/blob/master/doc/guides/test/ztest.rst)

jenmwms, 2020-09-01:
For the last two: `samples/subsys/ipc/.../remote`
My understanding is these do not need a `sample.yaml` here in the structure. The inclusion of `/remote` occurs in `samples/subsys/ipc/ipm_mcux/CMakeLists.txt ` and `samples/subsys/ipc/openamp/CMakeLists.txt`. For example:
```
include(ExternalProject)

ExternalProject_Add(
  openamp_remote
  SOURCE_DIR ${APPLICATION_SOURCE_DIR}/remote
```

So, only 2 of the 6 identified need a `sample.yaml`:
- samples/boards/intel_s1000_crb/dmic
- samples/subsys/mgmt/osdp

Please see the PR #27939

pabigot, 2020-09-02:
Thanks, I removed the `ipc` ones, and the out-of-tree where the `sample.yaml` is a couple directories up.

It's confusing that we have a test in the samples directory (where it probably isn't being executed), or maybe it's a non-executable sample ("example" would be better).  I'll just document it above and let it go....

PR #27913 (2020-09-03): samples: add missing sample.yaml files
Some samples were without sample.yaml. This adds
them.

Fixes #27813
commit e7eb428b (2020-09-03): samples: add missing sample.yaml files

Some samples were without sample.yaml. This adds
them.

Fixes #27813
closed 2020-09-03 by nashif