zephyrproject-rtos/zephyr #27819: Memory Management for MMU-based devices for LTS2

andrewboie, 2020-08-26:
Our existing memory protection design was designed for MPU-based systems. We now aim to introduce better support for user mode on systems with memory management hardware for the next Zephyr LTS version. This ticket is a meta-issue containing the overall plan of what we want this to look like.

We have already implemented MMU support on x86, but this is being overhauled to conform to this plan. For now, development effort will focus on having 32-bit x86 be the reference implementation to demonstrate features and provide an example for porting this to other architectures with MMUs.

The plan is to deliver all of this on x86 32-bit for LTS2.

**Design Goals**
We have a basic set of goals which inform how this plan came together:

- We want a single view of memory from the kernel's perspective.
  - This greatly simplifies debugging as pointer values mean the same thing regardless of what thread is scheduled, we would not need separate GDB inferiors for different address spaces.
  - This also makes concerns in #25950 go away since ISRs will be able to access all memory.
- We do not want a fundamentally different programming model for devices with an MPU or an MMU
  - Applications that support user mode should be portable between MPU and MMU devices
- We want to support devices with an MMU that are RAM constrained
  - We need to minimize the amount of memory used for page tables
  - If there are trade-offs with respect to real-time latency and memory usage, the developer must have a choice
- We need to minimize or eliminate the number of run-time memory allocations we have to make
  - This is greatly desirable from a functional safety perspective
  - If we do have to allocate, it must be clearly documented where and when this happens
- On a fundamental level, Zephyr is a micro-controller OS. We are not trying to re-implement Linux.

**Memory Model**

We now support arbitrary memory mapping now that #27001 has landed, we are no longer constrained by having to use strictly identity-mapped page tables. The kernel has a memory mapping API, but there are some important differences from how mmap() works, and the address space layout, in Linux:

- Mappings with respect to supervisor mode are the same regardless of what thread is running. 
  - If multiple sets of page tables exist, they all get updated by mapping calls.
  - If multiple sets of page tables exist, the only difference between page tables is access permissions with respect to user mode.
- The memory domain APIs remain the API for application memory access management.
  - Currently, memory domain partitions can only be within the linear address bounds of the permanent mappings for system RAM, but this will be changed to allow memory partitions for any mapped memory region.
  - Some improvements are planned or in code review for memory domains: #24609 #21991 #17833 #27786 
- The kernel's memory mapping function `k_mem_map()` isn't intended for application use at this time. It will soon be demoted to `z_mem_map()`.
- We can implement forms of ASLR for memory mappings
- The total bounds of the address space can be specified in Kconfig to establish an upper bound on the size of paging structures needed.

We will use memory mapping for the following use-cases:
- Memory mapping device driver MMIO regions with `device_map()`. This is already implemented.
- Optionally memory-mapping stacks
  - Provide virtual (instead of physical) guard pages for catching stack overflows.
  - Optionally implementing pagefault handling for guard pages to dynamically increase stack sizes
  - Memory mapping stacks will require an un-mapping function as threads exit.
- Optionally memory mapping heaps with `sbrk()`.
  - Because we have a single view of memory, heap locations will be managed at the memory domain level. We can extend this to MPU systems as well.
- VSDO-like pages to expose certain kernel global data to user mode in a read-only mapping, so that user mode can get this information without making system calls

By having the memory mapping done behind the scenes by the kernel, and using the memory domain APIs as the access management method for MMU devices, we present a single interface to applications regardless of the underlying hardware.

Memory mapping calls are one of the spots where an allocation may be required to instantiate new paging structures.
 - These allocations, once made, are permanent
 - We will provide a facility to optionally pre-allocate paging structures for the entire kernel address space (which is a subset of the full virtual address space whose bounds are controlled by Kconfig) at early boot to avoid any memory mapping allocations later.

**Page Table Management**
If `CONFIG_USERSPACE` is not enabled, then we just need one global set of page tables.

Currently, x86 has a partial set of page tables (covering the bounds of SRAM) for every user thread in the system. This uses way too much memory. Instead, we want to support two schemes of page table management, which can be chosen via Kconfig:

Option 1: Page tables at memory domain level
- A set of page tables will be maintained on a per memory domain basis. This is much less memory than page tables per-thread.
- Initializing a new memory domain will require allocations for new paging structures.
- Context switching is a fast machine register update to the page tables for the incoming thread's memory 
domain
  - If the address space is small enough, update links to leaf paging structures instead of top-level.
- Memory mapping calls or demand paging page-ins/outs will require all active page tables to be updated
- SMP systems will need some TLB shootdown code as the same page tables could be active on multiple CPUs
- Threads in the same memory domain will be able to access each other's stacks
  - As discussed in #25106
  - Portable code shouldn't assume such access

Option 2: Single set of page tables
- There is just one set of page tables per CPU. This uses the smallest amount of memory.
- Initializing a new memory domain will never require allocations.
- Context switching becomes slower and less deterministic
  - An O(n) operation to walk page tables, de-configure PTEs from the outgoing memory domain, and adjust PTEs for the incoming memory domain
  - This will mess with RTOS latency guarantees, although we should investigate if any of this can be done with irqs enabled

Demand paging (see #26088) will be supported in both configurations. Option 2 is for people who need memory usage to be absolutely minimized and don't care about real-time guarantees (they need a small OS, not necessarily an RTOS).

Nearly all the code for this will be at the `arch/` level. We plan to implement both options for 32-bit x86, and just option 1 for 64-bit.

andrewboie, 2020-10-28:
With https://github.com/zephyrproject-rtos/zephyr/pull/29411 we are very close. Remaining stories for x86 we are hoping to have for LTS2 if not much sooner:

- Demand paging: https://github.com/zephyrproject-rtos/zephyr/issues/26088
- Isolation for C libc globals: https://github.com/zephyrproject-rtos/zephyr/issues/25891
- VDSO page: no GH issue yet
- Anonymous page pool: https://github.com/zephyrproject-rtos/zephyr/issues/29526
- Memory un-mapping: https://github.com/zephyrproject-rtos/zephyr/issues/28900
- Memory-mapped thread stacks: https://github.com/zephyrproject-rtos/zephyr/issues/28899
- Finish sys_mutex, now that we have TLS https://github.com/zephyrproject-rtos/zephyr/issues/15138 https://github.com/zephyrproject-rtos/zephyr/issues/29520
- Arbitrary futex pointers: https://github.com/zephyrproject-rtos/zephyr/issues/25497

maksimmasalski, 2020-11-16:
> Demand paging (see #26088) will be supported in both configurations. Option 2 is for people who need memory usage to be absolutely minimized and don't care about hard real-time guarantees (they need a small embedded OS, not necessarily an RTOS).
> 
I think, if we will start implement features that are not aligned with RTOS itself, that can make Zephyr setup very messy. Is it better just to keep only option 1?

andrewboie, 2020-11-16:
>  Is it better just to keep only option 1?

No, for reasons that have already been enumerated.

> that can make Zephyr setup very messy. 

I have no idea what you mean here. It is one Kconfig.

closed 2022-11-18 by nashif