zephyrproject-rtos/zephyr #27820: reel_board:running failed in tests/drivers/gpio/gpio_api_1pin/

Zhaoningx, 2020-08-27:
Describe the bug
Running test cases and the output assert failed on reel_board

To Reproduce
Steps to reproduce the behavior:
west build -b reel_board tests/drivers/gpio/gpio_api_1pin/
west flash
see error

*** Booting Zephyr OS build zephyr-v2.3.0-2162-g46def95f510e  ***
Running test suite gpio_api_1pin_test

===================================================================
START - test_gpio_pin_configure_push_pull
Running test on port=GPIO_0, pin=11

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/gpio/gpio_api_1pin/src/test_config.c:23: pin_get_raw_and_verify: (val_expected )
Test point 10: invalid pin get value
 FAIL - test_gpio_pin_configure_push_pull

===================================================================
START - test_gpio_pin_configure_single_ended
Running test on port=GPIO_0, pin=11
When configured as input test pin value is high
 PASS - test_gpio_pin_configure_single_ended

===================================================================
START - test_gpio_pin_set_get_raw
Running test on port=GPIO_0, pin=11
 PASS - test_gpio_pin_set_get_raw

===================================================================
START - test_gpio_pin_set_get
.
.
.

Environment (please complete the following information):

OS: Fedora28
Toolchain: Zephyr-sdk-0.11.3
Commit ID: 94721b316f

pabigot, 2020-08-27:
Confirmed failure, but I suspect this is a test failure.  The Nordic GPIO driver has been pretty heavily tested on other platforms.

Zhaoningx, 2020-08-28:
Hello @pabigot , I don't get your point, you mean, this failure is normal and could be ignored on reel_board?

pabigot, 2020-08-28:
No, I mean this test depends on an input that is wired to an LED behaving in a particular way, which might not be how the circuit on this particular board actually behaves.  It doesn't mean the GPIO driver is wrong, though that is remotely possible.

Zhaoningx, 2020-08-28:
Good to know, so what should we do next?

pabigot, 2020-08-28:
Wait for the people who have been assigned this issue to look at it.

carlescufi, 2020-09-08:
@jfischer-phytec-iot would you mind taking a look at why this test is failing? It doesn't need any fixtures, just an LED.

jfischer-phytec-iot, 2020-09-09:
The LED (junction capacitance) and resistor between the GPIO pin and VDD form an RC element. The resistors on the board are intentionally larger, one for LED1 is the largest. The state of the pin is not yet settled at the time it is read, see:
_"Verify that it is not possible to change value of a pin via gpio_pin_set_raw function if pin is configured as an input."_ [gpio_pin_configure(port, TEST_PIN, GPIO_INPUT)](https://github.com/zephyrproject-rtos/zephyr/blob/master/tests/drivers/gpio/gpio_api_1pin/src/test_config.c#L119)
With the change below the test runs through. I am not sure if the value TEST_GPIO_MAX_RISE_FALL_TIME_US is always sufficient.

```diff
diff --git a/tests/drivers/gpio/gpio_api_1pin/src/test_config.c b/tests/drivers/gpio/gpio_api_1pin/src/test_config.c
index fa69e29a7e..0df0b56ea9 100644
--- a/tests/drivers/gpio/gpio_api_1pin/src/test_config.c
+++ b/tests/drivers/gpio/gpio_api_1pin/src/test_config.c
@@ -118,6 +118,7 @@ void test_gpio_pin_configure_push_pull(void)
         */
        ret = gpio_pin_configure(port, TEST_PIN, GPIO_INPUT);
        zassert_equal(ret, 0, "Failed to configure the pin as an input");
+       k_busy_wait(TEST_GPIO_MAX_RISE_FALL_TIME_US);
 
        int pin_in_val;

```

pabigot, 2020-09-09:
@jfischer-phytec-iot Thanks; I can work with that and will add that as an optional constant in the test's devicetree binding.  What delay value are you using here?

pabigot, 2020-09-09:
Ah, I see, it's defined in a header.  It should be in devicetree.

PR #28224 (2020-09-09): tests: gpio_api_1pin: fix failing test on reel_board
Need to add a rise/fall delay after configuring the pin as an input,
due to a long RC time constant on LED1 circuit.

Fixes #27820
closed 2020-09-09 by ioannisg