zephyrproject-rtos/zephyr #27821: frdm_k64f:running test cases error

Zhaoningx, 2020-08-27:
Describe the bug
running test case /tests/subsys/power/power_mgmt/ error, make system crash.

To Reproduce
Steps to reproduce the behavior:

west build -b frdm_k64f tests/subsys/power/power_mgmt/
west flash
see error

*** Booting Zephyr OS build zephyr-v2.3.0-2118-g94721b316f63  ***
Running test suite power_management_test

===================================================================
START - test_power_idle
give way to idle thread
 PASS - test_power_idle

===================================================================
START - test_power_state_trans

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/power/power_mgmt/src/main.c:E: ***** BUS FAULT *****
E:   Stacking error
E:   Imprecise data bus error
E:   NXP MPU error, port 3
E:     Mode: Supervisor, Data Address: 0x20001110
E:     Type: Write, Master: 0, Regions: 0x8200
E: r0/a1:  0x00000000  r1/a2:  0x00001b55  r2/a3:  0x200015b4
E: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x00000000
E:  xpsr:  0x00000000
E: Faulting instruction address (r15/pc): 0x00000000
E: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
E: Current thread: 0x200007c8 (idle 00)
E: Halting system

Environment (please complete the following information):

OS: Fedora28
Toolchain: Zephyr-sdk-0.11.3
Commit ID: g94721b316f63

nashif, 2020-08-27:
@hakehuang  @dleach02 FYI

hakehuang, 2020-08-28:
same situation here
```
Running test suite power_management_test
===================================================================
START - test_power_idle
give way to idle thread
 PASS - test_power_idle
===================================================================
START - test_power_state_trans

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/power/power_mgmt/src/main.c:E: ***** BUS FAULT *****
E:   Stacking error
E:   Imprecise data bus error
E:   NXP MPU error, port 3
E:     Mode: Supervisor, Data Address: 0x20001110
E:     Type: Write, Master: 0, Regions: 0x8200
E: r0/a1:  0x00000000  r1/a2:  0x00001b55  r2/a3:  0x00000000
E: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x00000000
E:  xpsr:  0x00000000
E: Faulting instruction address (r15/pc): 0x00000000
E: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
E: Current thread: 0x200007c8 (idle 00)
E: Halting system

================================ End =================================
```

and all our drivers using the device_pm_control_nop controler.

dleach02, 2020-08-31:
Folks, The idle stack is not big enough so you end up with the fault occurring when trying to print an assert message. This actually hides your real problem with this test on this board.

Add `CONFIG_IDLE_STACK_SIZE=640` to the prj.conf file. 

No when you run the tests you will see the following error:

Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/power/power_mgmt/src/main.c:45: sys_set_power_state: (device_power_state == DEVICE_PM_ACTIVE_STATE is true)

ASSERTION FAIL [!(thread->base.user_options & ((1UL << (0))))] @ WEST_TOPDIR/zephyr/arch/arm/core/aarch32/cortex_m/thread_abort.c:35
	essential thread aborted
E: r0/a1:  0x00000004  r1/a2:  0x00000023  r2/a3:  0x00000080
E: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x0000220d
E:  xpsr:  0x61000000
E: Faulting instruction address (r15/pc): 0x00006410
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x200007c8 (idle 00)
E: Halting system

I added some debug and can tell that the dummy function for suspend is never called. I'm not familiar with device power management so I'm curious if this test works with any additional boards?

hakehuang, 2020-08-31:
@dleach02 only frdm_k64f is enabled for this test application for NXP platform.

dleach02, 2020-08-31:
I'm not familiar with the power management subsystem. This is a brand new test so it isn't clear to me if there is an underlying failure or not. Someone familiar with power management will need to look at this to ensure that 1) the test is structured correctly and 2) that power management is actually implemented for the k64f.

carlescufi, 2020-09-08:
@dleach02 remove the board from the allowlist

hakehuang, 2020-09-09:
@dleach02 v2.4.0-rc1 still not removed

maksimmasalski, 2020-09-17:
@dleach02 What is your solution for that task? Seems just necessary to remove board from the allowlist.

PR #28677 (2020-09-24): test: power: Remove frdm_k64f platform from allowed list
Removing frdm_k64f from the allowed list of platforms to test.

Fixes: #27821
commit b94dbc56 (2020-09-24): test: power: Remove frdm_k64f platform from allowed list

Removing frdm_k64f from the allowed list of platforms to test.
Power management support has not been added and/or verified for
frdm_k64f platform yet.

Fixes: #27821
closed 2020-09-24 by MaureenHelm