zephyrproject-rtos/zephyr #27832: [Coverity CID :212957] Out-of-bounds access in arch/x86/core/x86_mmu.c

zephyrbot, 2020-08-27:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/63e858262936c303ccb334ba6db680c3cd135823/arch/x86/core/x86_mmu.c#L748

Category: Memory - corruptions
Function: `stack_guard_set`
Component: Architectures
CID: [212957](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=212957)

Details:

```
742    
743         assert_virt_addr_aligned(guard_page);
744    
745         /* Always modify the kernel's page tables since this is for
746          * supervisor threads or handling syscalls
747          */
>>>     CID 212957:  Memory - corruptions  (ARRAY_VS_SINGLETON)
>>>     Passing "&z_x86_kernel_ptables" to function "page_map_set" which uses it as an array. This might corrupt or misinterpret adjacent memory locations.
748         ret = page_map_set(&z_x86_kernel_ptables, guard_page, pte,
749                    page_pool_get, NULL);
750         /* Literally should never happen */
751         __ASSERT(ret == 0, "stack guard mapping failed for %p", guard_page);
752         (void)ret;
753     }
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

ceolin, 2020-09-17:
This really looks like a false positive, the memory pointed by ptables is allocated by the linker. Everything looks fine IMHO

wentongwu, 2020-09-21:
@ceolin Thanks, I just get time to look at this, but can we make z_x86_kernel_ptables like this `extern pentry_t z_x86_kernel_ptables[];`?

PR #28557 (2020-09-22): arch: x86: mmu: use z_x86_kernel_ptables as array.
Use z_x86_kernel_ptables as array to make Coverity happy.

Coverity-CID: 212957.
Coverity-CID: 212958.

Fixes: #27832 
Fixes: #27833
closed 2020-09-22 by MaureenHelm