zephyrproject-rtos/zephyr #27833: [Coverity CID :212958] Out-of-bounds access in arch/x86/core/x86_mmu.c

zephyrbot, 2020-08-27:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/63e858262936c303ccb334ba6db680c3cd135823/arch/x86/core/x86_mmu.c#L679

Category: Memory - corruptions
Function: `arch_mem_map`
Component: Architectures
CID: [212958](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=212958)

Details:

```
673    
674         for (size_t offset = 0; offset < size; offset += CONFIG_MMU_PAGE_SIZE) {
675             int ret;
676             pentry_t entry_val = (phys + offset) | entry_flags;
677             uint8_t *dest_virt = (uint8_t *)virt + offset;
678    
>>>     CID 212958:  Memory - corruptions  (ARRAY_VS_SINGLETON)
>>>     Passing "ptables" to function "page_map_set" which uses it as an array. This might corrupt or misinterpret adjacent memory locations.
679             ret = page_map_set(ptables, dest_virt, entry_val,
680                        page_pool_get, NULL);
681    
682             /* Currently used for new mappings, no TLB flush. Re-visit
683              * as capabilities increase
684              */
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

ceolin, 2020-09-17:
This really looks like a false positive, the memory pointed by `ptables` is allocated by the linker. Everything looks fine IMHO

wentongwu, 2020-09-21:
@ceolin Thanks, I just got time to look at this, but can we make z_x86_kernel_ptables like this `extern pentry_t z_x86_kernel_ptables[];`?

PR #28557 (2020-09-22): arch: x86: mmu: use z_x86_kernel_ptables as array.
Use z_x86_kernel_ptables as array to make Coverity happy.

Coverity-CID: 212957.
Coverity-CID: 212958.

Fixes: #27832 
Fixes: #27833
closed 2020-09-22 by MaureenHelm