zephyrproject-rtos/zephyr #27834: [Coverity CID :212959] Macro compares unsigned to 0 in include/sys/mem_manage.h

zephyrbot, 2020-08-27:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/63e858262936c303ccb334ba6db680c3cd135823/include/sys/mem_manage.h#L150

Category: Integer handling issues
Function: `z_mem_virt_addr`
Component: Other
CID: [212959](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=212959)

Details:

```
144         return Z_MEM_PHYS_ADDR(addr);
145     }
146    
147     /* Just like Z_MEM_VIRT_ADDR() but with type safety and assertions */
148     static inline void *z_mem_virt_addr(uintptr_t phys)
149     {
>>>     CID 212959:  Integer handling issues  (NO_EFFECT)
>>>     This greater-than-or-equal-to-zero comparison of an unsigned value is always true. "phys >= 0UL".
150         __ASSERT((phys >= CONFIG_SRAM_BASE_ADDRESS) &&
151              (phys < (CONFIG_SRAM_BASE_ADDRESS +
152                   (CONFIG_SRAM_SIZE * 1024UL))),
153              "physical address 0x%lx not in RAM", (unsigned long)phys);
154    
155         return (void *)Z_MEM_VIRT_ADDR(phys);
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

wentongwu, 2020-09-09:
CONFIG_SRAM_BASE_ADDRESS is not guaranteed to be 0

de-nordic, 2020-09-09:
@wentongwu Marked as "False Positive" with Action set to "Ignore"

wentongwu, 2020-09-09:
Thanks @de-nordic

closed 2020-09-09 by wentongwu