zephyrproject-rtos/zephyr #27835: [Coverity CID :212962] Macro compares unsigned to 0 in include/sys/mem_manage.h

zephyrbot, 2020-08-27:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/63e858262936c303ccb334ba6db680c3cd135823/include/sys/mem_manage.h#L132

Category: Integer handling issues
Function: `z_mem_phys_addr`
Component: Other
CID: [212962](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=212962)

Details:

```
126     /* Just like Z_MEM_PHYS_ADDR() but with type safety and assertions */
127     static inline uintptr_t z_mem_phys_addr(void *virt)
128     {
129         uintptr_t addr = (uintptr_t)virt;
130    
131     #ifdef CONFIG_MMU
>>>     CID 212962:  Integer handling issues  (NO_EFFECT)
>>>     This greater-than-or-equal-to-zero comparison of an unsigned value is always true. "addr >= 0UL".
132         __ASSERT((addr >= CONFIG_KERNEL_VM_BASE) &&
133              (addr < (CONFIG_KERNEL_VM_BASE + (CONFIG_SRAM_SIZE * 1024UL))),
134              "address %p not in permanent mappings", virt);
135     #else
136         /* Should be identity-mapped */
137         __ASSERT((addr >= CONFIG_SRAM_BASE_ADDRESS) &&
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

andrewboie, 2020-08-27:
false positive
CONFIG_KERNEL_VM_BASE is not guaranteed to  be 0

wentongwu, 2020-09-09:
agree, CONFIG_KERNEL_VM_BASE is not guaranteed to be 0

de-nordic, 2020-09-09:
@wentongwu Marked as "False Positive" with Action set to "Ignore".

closed 2020-09-09 by wentongwu