zephyrproject-rtos/zephyr #27841: samples: disk: unable to access fat fs on sd card

FRASTM, 2020-08-27:
**Describe the bug**
The **samples/subsys/fs/fat_fs/**  application cannot access the SD card through SPI bus, 

**To Reproduce**
This sample is using a ci_shield to mount a SD card through SPI on a nucleo stm32f411

$ west build -p auto -b nucleo_f411re samples/subsys/fs/fat_fs/ -DSHIELD="ci_shie ld"

See error: 

`Error mounting disk.          `
`[00:00:00.007,000] <err> main: Storage init ERROR!             `
`[00:00:00.009,000] <err> fs: fs mount error (-5)`

**Expected behavior** 
`[00:00:00.005,000] <inf> spi_ll_stm32: CS control inhibited (no GPIO device)    
Sector size 512                                                                 
Memory Size(MB) 1886                                                            
Disk mounted.                                                                   
                                                                                
Listing dir /SD: ...                                                            
[DIR ] SYSTEM~1                                                                 
[FILE] TEST.LOG (size = 3072000)                                                
[DIR ] TEST                                                                     
`

FRASTM, 2020-08-28:
The second `sdhc_spi_detect` fails, in the `disk_spi_sdhc_access_init` is returning the error code -5

(This is not the case when reverting  commit c64e0ce (from PR #27685) )

de-nordic, 2020-08-28:
@FRASTM I  do not think that this is problem with FAT FS.

FRASTM, 2020-08-28:
yes, at the beginning the pb occurs with the sample _fat_fs_,

de-nordic, 2020-08-28:
> 
> 
> yes, at the beginning the pb occurs with the sample _fat_fs_,

Thanks for clarification! So the issue probably is not even with the samples but the general SD access via SPI.

de-nordic, 2020-08-28:
This is probably similar issue: https://github.com/zephyrproject-rtos/zephyr/issues/27112

henrikbrixandersen, 2020-08-30:
> This is probably similar issue: #27112

Unlikely since the author of that issue report talks about Zephyr 2.3.0, which does not contain c64e0ce923cde54415d203e447a1dc62894ed2b0.

henrikbrixandersen, 2020-08-30:
@FRASTM Does the SPI controller driver used for accessing the SDHC card support `SPI_HOLD_ON_CS` properly? The Zephyr SPI SDHC disk access driver heavily depends on that.

FRASTM, 2020-08-31:
thank you  @henrikbrixandersen for pointing that. Trying CONFIG_SPI_HOLD_ON_CS=y gives not the solution, though. Could you please describe a bit the way for the spi driver to support `SPI_HOLD_ON_CS`.

henrikbrixandersen, 2020-08-31:
@FRASTM `SPI_HOLD_ON_CS` is not a Kconfig option. It is a SPI transceive operation option: https://github.com/zephyrproject-rtos/zephyr/blob/36aa42f384382fd71f0e3ebdd33cd1bd434577e1/include/drivers/spi.h#L104

FRASTM, 2020-09-01:
well, the `sdhc_spi_init` in the disk_access_spi_sdhc.c has :

> data->cfg.operation = SPI_WORD_SET(8) | SPI_HOLD_ON_CS;

carlescufi, 2020-09-08:
@erwango @FRASTM could you please debug this a bit further to see if it's related to the particular SPI driver?

NZSmartie, 2020-09-09:
Also wanting to confirm a similar issue while giving `2.4.0-rc1` with a Nordic `nRF52840` a try. The SD card does not mount:
```
[00:00:00.106,536] <err> fs: fs mount error (-5)
```

When reverting c64e0ce923cde54415d203e447a1dc62894ed2b0, it works again

```
[00:00:00.505,218] <inf> sdhc_spi: Found a ~982 MiB SDHC card.
[00:00:00.506,469] <inf> sdhc_spi: Manufacturer ID=2 OEM='TM' Name='SD01G' Revision=0x28 Serial=0x9e50e185
```

The DTS for the SPI is:
```dts
&spi1 {
	compatible = "nordic,nrf-spi";
	status = "okay";
	sck-pin = <29>;
	miso-pin = <31>;
	mosi-pin = <30>;

	cs-gpios = <&gpio0 16 GPIO_ACTIVE_HIGH>;

	sdcard: sdcard@0 {
		compatible = "zephyr,mmc-spi-slot";
		reg = <0>;
		status = "okay";
		label = "SDCARD";
		spi-max-frequency = <24000000>;
	};
};
```

henrikbrixandersen, 2020-09-09:
@pabigot You mentioned that you verified commit c64e0ce923cde54415d203e447a1dc62894ed2b0 on Nordic SPI in https://github.com/zephyrproject-rtos/zephyr/pull/27685#pullrequestreview-471616998. Any idea on what goes wrong for @NZSmartie above?

NZSmartie, 2020-09-09:
I'll have a dig/debug into the situation tomorrow. I've been looking into the nrfx spi driver and so far it looks to respect `SPI_HOLD_ON_CS` indirectly by `spi_context_cs_control()`
https://github.com/zephyrproject-rtos/zephyr/blob/016b5422d90e9d3405eda22da4fda23d6a13a415/drivers/spi/spi_context.h#L176-L200
from here
https://github.com/zephyrproject-rtos/zephyr/blob/016b5422d90e9d3405eda22da4fda23d6a13a415/drivers/spi/spi_nrfx_spi.c#L165

pabigot, 2020-09-09:
I'm not sure what board's being used in https://github.com/zephyrproject-rtos/zephyr/issues/27841#issuecomment-689554272 because the nrf52840dk_nrf52840 has a different `spi1` configuration.

A clear problem in @NZSmartie's DTS is specifying the chip select as active-high.  CSn should be active-low.

What works for me is this:
```
tirzah[22]$ cat boards/nrf52840dk_nrf52840.conf
CONFIG_LOG=y
CONFIG_DISK_ACCESS_SDHC=y
CONFIG_DISK_ACCESS_SPI_SDHC=y
CONFIG_SPI=y
tirzah[23]$ cat boards/nrf52840dk_nrf52840.overlay
#if 1
&arduino_spi {
  status = "disabled";
};

&spi2 { // nordic,nrf-spi
  status = "okay";
  sck-pin = <47>;
  miso-pin = <46>;
  mosi-pin = <45>;
  cs-gpios = <&arduino_header 16 GPIO_ACTIVE_LOW>; /* D10 */
#else
&arduino_spi { // = spi3 : nordic,nrf-spim
#endif
  sdhc0: sdhc@0 {
    compatible = "zephyr,mmc-spi-slot";
    reg = <0>;
    status = "okay";
    label = "SDHC0";
    spi-max-frequency = <100000>;
  };
};
```

Note I'm switching from spi3 to spi2 because spi3 (which only supports `nordic,nrf-spim`) doesn't work, and it didn't work before #27685 either.  Looks like CSn is asserted except for a very brief period right before the transaction begins.  (#27685 clearly took a step towards fixing SPIM, since before that patch CSn was inverted relative to its behavior afterwards.  But SPIM doesn't deassert CSn after the transaction.)

NZSmartie, 2020-09-10:
> A clear problem in @NZSmartie's DTS is specifying the chip select as active-high. CSn should be active-low.

Oh hmm. I haven't paid too much attention to the SPI CS line, their spec is rather vague about the pin itself[1]. I was following the Device tree example set by most boards and most but one that specify `GPIO_ACTIVE_HIGH`. 
Taking the _OLIMEXINO-STM32_ as an example. [the schematic](https://www.olimex.com/Products/Duino/STM32/OLIMEXINO-STM32/resources/OLIMEXINO-STM32_sch_latest.pdf) pulls the CS line low with a 1M resistor, I guess I need to read the MMC spec as well. 
https://github.com/zephyrproject-rtos/zephyr/blob/c6b1375400f1bea7acdadd123c5492d8985c1fd8/boards/arm/olimexino_stm32/olimexino_stm32.dts#L80-L91
It still doesn't explain why `GPIO_ACTIVE_HIGH` works reliably in `2.3.0`

Another interesting note. After setting `cs-gpios = <&gpio0 16 GPIO_ACTIVE_LOW>;` and using a Sandisk 32GB SDHC. It works without problems:
```
[00:00:00.170,166] <inf> sdhc_spi: Found a ~30436 MiB SDHC card.
[00:00:00.171,478] <inf> sdhc_spi: Manufacturer ID=3 OEM='SD' Name='SE32G' Revision=0x80 Serial=0x3c540a16
```
I guess the OEM SD card has some quirks that was compatible with the previous CS handling?

---

[1] _Physical Layer Simplified Specification Version 7.10_ - _7.2.1 Mode Selection and Initialization_
> The SD Card is poweredup in the SD mode. It will enter SPI mode if the CS signal is asserted (negative) during the reception of the reset command (CMD0).

NZSmartie, 2020-09-10:
Also, the OEM SDCard times out during initialisation with `GPIO_ACTIVE_LOW` I'm still checking it out. It's a bit hard to get a logic analyser on the signals in my board, It's already a bodge job on a prototype, and don't have any test pads spare 😅

pabigot, 2020-09-10:
In Zephyr 2.3.0 the active level flags from devicetree were ignored, and the SPI infrastructure set the configuration at runtime, defaulting to active low.  ("asserted (negative)" in the spec you quoted means active low; active high would be "asserted (positive)".)  Note that most non-SPI uses of GPIO are indeed active-high.

It could well be the card, there are many standards and classes.  I've had to go back to class 4 cards from five years ago to get some interfaces to work.  In any case all I can confirm is that on current master the sample works correctly on Nordic boards when using the `nordic,nrf-spi` driver.

pabigot, 2020-09-10:
It appears that, if `CONFIG_SPI_STM32_USE_HW_SS=y`, STM32 driver is not respecting `SPI_HOLD_ON_CS`: logic analyzer traces show CS is being deasserted between the transmission of an R1 command and the read of its result.  This produces unexpected results.

With software control of SSn this doesn't happen, but the response to the command is still not the expected result.

Barmaley13, 2020-09-25:
I'm pretty new to Zephyr but I am seeing similar error. 
```
Error mounting disk.
[00:00:00.006,000] [1;31m<err> main: Storage init ERROR![0m
[00:00:00.007,000] [1;31m<err> fs: fs mount error (-5)[0m
```

I was trying to run `fat_fs` sample on `nucleo_f411re` board.
Added following files as board overlays. 

nucleo_f411.conf
```
CONFIG_LOG=y
CONFIG_SPI_LOG_LEVEL=0
CONFIG_DISK_ACCESS_SDHC=y
CONFIG_DISK_ACCESS_SPI_SDHC=y
CONFIG_SPI=y
CONFIG_SPI_STM32=y
# CONFIG_SPI_STM32_INTERRUPT=y
CONFIG_SPI_1=y

CONFIG_DISK_ACCESS=y
CONFIG_FILE_SYSTEM=y
CONFIG_FAT_FILESYSTEM_ELM=y
```
and nucleo_f411re.overlay
```
&spi1 {
	cs-gpios = <&gpiob 6 (GPIO_ACTIVE_LOW | GPIO_PULL_UP)>; /* D10 */

        sdhc0: sdhc@0 {
                compatible = "zephyr,mmc-spi-slot";
                reg = <0>;
                status = "okay";
                label = "SDHC0";
                spi-max-frequency = <24000000>;
	};
};
```
I just wired wrapped SD card breakout board to spi1 and using D10 as a chip select. It seems like SPI bus does not work properly. I was not able to trigger it at all on `v2.4.0-rc2`. I was able to trigger spi on v.2.2.0 but it does not look right...
[image]
miso looks like sck and cs looks like data... 

Would love to debug this further if someone can provide pointers.

closed 2020-12-09 by github-actions[bot]

PureEngineering, 2022-05-04:
I would like to repoen this issue. as still seeing this issue.